project-chip/connectedhomeip #18839: [TC-IDM-4.3]--Chip tool needs to provide a way to set MinIntervalFloor value > MinIntervalFloor value For subscription request.

Rajashreekalmane, 2022-05-26:
#### Problem
As per Test-Plan, step 20 says TH sends a subscription request action for attribute and sets the MinIntervalFloor value to be same as MaxIntervalCeiling. Activate the Subscription between TH and DUT. Modify the attribute which has been subscribed to on the DUT.
**Chip tool needs to provide a way to set MinIntervalFloor value > MinIntervalFloor value when TH sends a subscription request action for attribute to DUT.**
 
**Test-Plan screenshot:-**
[image]

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#subscribe-response-action

bzbarsky-apple, 2022-05-26:
@Rajashreekalmane @cjandhyala I don't understand this issue.  The test plan says to set MinIntervalFloor to the same value as MaxIntervalCeiling (which chip-tool supports), but the text talks about setting `MinIntervalFloor > MinIntervalFloor`, which is obviously not possible.

bzbarsky-apple, 2022-05-26:
I am _guessing_ this issue is actually about step 21, not step 20, and that the request is that we allow `MinIntervalFloor > MaxIntervalCeiling`?

If so, that's not allowed in the SDK right now, so adding it to chip-tool is not exactly trivial.

andy31415, 2022-05-26:
Are we supposed to cert-block on negative test cases? I imagine many of these may be non-trivial to implement.

@cjandhyala what is the policy here? I would be inclined to not block on "allow invalid values to be accepted by tooling" or at least separate these cases a bit, since we will want some flags/builds/error injection capability for "I am providing invalid values but I know what I am doing and really need it"

Rajashreekalmane, 2022-05-26:
@bzbarsky-apple Yes this issue is on step 21 , I have updated  the issue Now.

mrjerryjohns, 2022-05-26:
I don't think is validatable without modifications to the IM implementation in the SDK to permit this 'test modality' and skips initial validation of these parameters.

I'd recommend we skip this test.

bzbarsky-apple, 2022-05-27:
@mrjerryjohns See #18855

closed 2022-05-27 by bzbarsky-apple