project-chip/connectedhomeip #18848: [TC-CC-9.*] ColorLoopStoredEnhancedHue attribute value  mismatch error.

manjunath-grl, 2022-05-26:
Summary:
As per testplan [TC-CC-9.*] Precondition: Set DUT EnhancedCurrentHue to 0x4000 using EnhancedMoveToHue command
but in Step 5c: TH read ColorLoopStoredEnhancedHue attribute from DUT and value is the same as EnhancedCurrentHue attribute (0x4000) set in Pre-Condition Step #3 but got 0 when executed.
Testplan screenshot:
[image]
[image]
Execution logs:
[CLSEH_Value.txt](https://github.com/project-chip/connectedhomeip/files/8780028/CLSEH_Value.txt)
[image]

bzbarsky-apple, 2022-05-26:
@jmartinez-silabs

jmartinez-silabs, 2022-05-26:
I will review it but I think those are teat plan instructions set issues and not implementation issues

woody-apple, 2022-05-27:
SDK Spec Review: Tagging @cjandhyala, as this may be an issue with the test plan

jmartinez-silabs, 2022-06-03:
Spec and Test plan are correct. I tested manually following the test-plan instructions and the implementation matches the expected results of the test plan.

Here are all the logs of test case 9.1 I ran.
[TC-CC-9-1.txt](https://github.com/project-chip/connectedhomeip/files/8835662/TC-CC-9-1.txt)

ColorLoopStoredEnhancedHue gets updated to the EnhancedHue Value before a loop. 

I suspect the issue you faced is either a Yaml test issue or the pre-conditions for the test were not met.

Who is expected to fix/validate the YAML tests?

woody-apple, 2022-06-15:
Spec Review: Tagging @raju-apple to take a look at updating the YAML/tests as necessary.

raju-apple, 2022-06-16:
@manjunath-grl Can you re-try today again and see if its still happening and check.

manjunath-grl, 2022-06-21:
@woody-apple @raju-apple after Precondition we added a delay of 1 sec now it is working fine
YAML screenshot:
[image]
Execution logs:
[CC_9_1.zip](https://github.com/project-chip/connectedhomeip/files/8948176/CC_9_1.zip)

closed 2022-06-21 by manjunath-grl