project-chip/connectedhomeip #18850: [TC-CC-9.*] EnhancedCurrentHue value mismatch error.

manjunath-grl, 2022-05-26:
Summary:
As per testplan [TC-CC-9.*] TH sends ColorLoopSet command  EnhancedCurrentHue should decrease and continue to Max allowed enhanced hue if reach 0. After 30 sec, it should return to ColorLoopStartEnhancedHue but got random value within 0 to 65535 when executed.

Testplan screenshot:
[image]

Execution logs:
[ECH_random.txt](https://github.com/project-chip/connectedhomeip/files/8780091/ECH_random.txt)
[image]
[ECH_value.txt](https://github.com/project-chip/connectedhomeip/files/8780093/ECH_value.txt)
[image]

bzbarsky-apple, 2022-05-26:
@jmartinez-silabs

woody-apple, 2022-05-27:
SDK Spec Review: Tagging @cjandhyala to confirm that this is or is not a cert blocker, or a test plan issue.

cjandhyala, 2022-05-27:
@woody-apple ColorLoopSet is an optional command and we are not going support optional cmds for v1.0, hence not cert blocker

[image]

yinyihu-silabs, 2022-05-30:
@manjunath-grl 
The two values you read is normal. with color loop, the _EnhancedCurrentHue_ is continuously changing. Unless you're reading _EnhancedCurrentHue_ exactly 30 seconds from the time of activation (not the same as time between sending activation and read command), you're likely not going to received the exact match.

Based on the logs, I assumed you set the initial _EnhancedCurrentHue_ to 160. Both 379 and 65477 are reasonably within the range of 160 where 379 is likely just short of 30 seconds and 65477 is just over 30 seconds wrapping over the Max limit.

In Step 5e follows, it's expected that  _EnhancedCurrentHue_  will continue decreasing over every 30-second loop.

manjunath-grl, 2022-06-02:
@yinyihu-silabs thanks for the information

closed 2022-06-02 by manjunath-grl