project-chip/connectedhomeip #18864: Thread accessory (nordic board) does not boot up

phonnakasturi-apple, 2022-05-26:
With SHA eb7628f103faeb53b29519a6d40a47bafb290f62

Thread test fails because nordic board cannot be started.

**Steps:**
 1. cd ~/connectedhomeip
 2. git pull
 3. git reset --hard eb7628f103faeb53b29519a6d40a47bafb290f62
 4. source scripts/activate.sh
 5. python3 scripts/setup/nrfconnect/update_ncs.py --update
 6. cd examples/lighting-app/nrfconnect
 7. west build -b nrf52840dk_nrf52840 --pristine always
 8. west flash --erase
 9. minimum -s
 10. Set the port number (can be obtained by ls /dev/tty.*)
 
**Output**
minimum terminal hangs;

% minicom -s

Welcome to minicom 2.8

OPTIONS:                                                                     
Compiled on Oct 25 2021, 04:59:49.                                           
Port /dev/tty.usbmodem0006832436881, 13:28:31                                
                                                                             
Press Meta-Z for help on special keys                                        
                                                                             
                                                                             
Also
press + hold of button 1 does not reset the board as no led lights are getting flashed

kean-apple, 2022-05-27:
This is specific to Nordic example lighting app. Nordic all clusters app powers up ok

bzbarsky-apple, 2022-05-27:
@Damian-Nordic

phonnakasturi-apple, 2022-05-28:
WIth SHA f6a79db073621a2af0b3b5c5317fd8dc4028c58c
This issue was resolved.
Nordic lighting app was able to power up.

closed 2022-05-28 by phonnakasturi-apple