project-chip/connectedhomeip #18868: [Chip Tool Darwin] Cannot send group messages

dwgaul, 2022-05-26:
#### Attempting to send a group message using chip tool darwin:
 
`./chip-tool-darwin onoff on 0xffffffffffff4141 1`

But it looks like chip tool darwin is treating this address as a regular address and not going through the group message flow:

```
[1653606382458] [42989:13340274] CHIP: [DIS] Fabric (0x3) persisted to storage. Calling OnFabricPersistedToStorage
[1653606382458] [42989:13340274] CHIP: [CTL] Joined the fabric at index 3. Compressed fabric ID is: 0xA2BFDCEC680E682F
[1653606382458] [42989:13340274] CHIP: [TOO] Sending command to node 0xFFFFFFFFFFFF4141
[1653606382458] [42989:13340277] CHIP: [CSM] FindOrEstablishSession: PeerId = 20EBAACAD67DEC05:FFFFFFFFFFFF4141
[1653606382458] [42989:13340277] CHIP: [CSM] FindOrEstablishSession: No existing OperationalDeviceProxy instance found
[1653606382458] [42989:13340277] CHIP: [DIS] Resolving 20EBAACAD67DEC05:FFFFFFFFFFFF4141 ...
[1653606382458] [42989:13340277] CHIP: [DL] Resolve type=_matter._tcp name=20EBAACAD67DEC05-FFFFFFFFFFFF4141 interface=0
[1653606382660] [42989:13340278] CHIP: [DIS] Checking node lookup status after 201 ms
[1653606383145] [42989:13340277] CHIP: [DL] Mdns: OnRegister name: 20EBAACAD67DEC05-195DD004278F56C2, type: _matter._tcp., domain: local., flags: 2
[1653606383145] [42989:13340277] CHIP: [DIS] mDNS service published: _matter._tcp
[1653606383145] [42989:13340277] CHIP: [DL] Mdns: OnRegister name: CC9D98434025D406-17436FA45CFBB716, type: _matter._tcp., domain: local., flags: 2
[1653606383145] [42989:13340277] CHIP: [DIS] mDNS service published: _matter._tcp
[1653606383145] [42989:13340277] CHIP: [DL] Mdns: OnRegister name: A2BFDCEC680E682F-69085F2CDD3C0480, type: _matter._tcp., domain: local., flags: 2
[1653606383145] [42989:13340277] CHIP: [DIS] mDNS service published: _matter._tcp
[1653606392460] [42989:13340274] CHIP: [-] ../../commands/common/CHIPCommandBridge.mm:122: CHIP Error 0x00000032: Timeout at ../../commands/common/CHIPCommandBridge.mm:81
[1653606392460] [42989:13340274] CHIP: [TOO] Run command failure: ../../commands/common/CHIPCommandBridge.mm:122: CHIP Error 0x00000032: Timeout
```

Meanwhile the standard chip tool looks to go through a group message flow, sending the command to a group:

```
./chip-tool onoff on 0xffffffffffff4141 1
...
[1653606368947] [42987:13340210] CHIP: [DIS] Added new fabric at index: 0x6, Initialized: 1
[1653606368947] [42987:13340210] CHIP: [DIS] Assigned compressed fabric ID: 0xC216416E51CC7AEA, node ID: 0x000000000001B669
[1653606368954] [42987:13340210] CHIP: [DIS] Fabric (0x6) persisted to storage. Calling OnFabricPersistedToStorage
[1653606368954] [42987:13340210] CHIP: [CTL] Joined the fabric at index 6. Compressed fabric ID is: 0xC216416E51CC7AEA
[1653606368954] [42987:13340210] CHIP: [IN] UDP::Init bind&listen port=5550
[1653606368954] [42987:13340210] CHIP: [IN] UDP::Init bound to port=5550
[1653606368954] [42987:13340210] CHIP: [IN] UDP::Init bind&listen port=5550
[1653606368954] [42987:13340210] CHIP: [IN] UDP::Init bound to port=5550
[1653606368954] [42987:13340210] CHIP: [IN] TransportMgr initialized
[1653606368993] [42987:13340212] CHIP: [DL] _OnPlatformEvent
[1653606368993] [42987:13340212] CHIP: [TOO] Sending command to group 0x4141
[1653606368993] [42987:13340212] CHIP: [TOO] Sending cluster (0x00000006) command (0x00000001) on Group 16705
```

Is group message support intended to be included in chip tool darwin? Right now we would be unable to run certification tests for group messaging via the chip tool darwin app.

bzbarsky-apple, 2022-05-27:
> Is group message support intended to be included in chip tool darwin?

Not unless and until the Darwin framework starts sending group messages.  See also https://github.com/project-chip/connectedhomeip/pull/18375

>  Right now we would be unable to run certification tests for group messaging via the chip tool darwin app.

Which tests require sending of group messages that you're trying to run via chip-tool-darwin?  In general, certification can't require a DUT to send group messages if it just doesn't send them....

dwgaul, 2022-05-27:
> Not unless and until the Darwin framework starts sending group messages. See also https://github.com/project-chip/connectedhomeip/pull/18375

Ok good to know. Asked around and it sounded like this was the case, but good to get official confirmation.

> Which tests require sending of group messages that you're trying to run via chip-tool-darwin? In general, certification can't require a DUT to send group messages if it just doesn't send them....

We're running through our internal TE9 testing and I've been assigned a few group related test cases:

`TC-SC-5.2 Receiving a group message`
I think this one maybe considered N/A for us. Unless my understanding is incorrect, I don't believe clients would receive group messages?  

`TC-SC-5.3 Sending a group message`
This is the main one we were trying to validate. I'll look into what the expectation is for being required in certification, at a quick glance I don't see anything in the spec around group messaging being optional or mandatory.

bzbarsky-apple, 2022-05-28:
Right, the "sending a group message" testcase can only be run on clients that would actually decide to send group messages....  That all makes sense.