project-chip/connectedhomeip #18875: In thermostat cluster app GetRelayStatusLog command throws (INVALID_COMMAND) error on chip tool + chip tool Darwin

phonnakasturi-apple, 2022-05-27:
The test case has 3 steps;

1. To increase the temperature by SetpointRaiseLower command
2. To Lower the temperature by SetpointRaiseLower command
3. To issue GetRelayStatusLog command to the Test Harness

The 3rd step that tests the GetRelayStatusLog throws (INVALID_COMMAND) error on chip tool + chip tool Darwin

**Note:** Accessory(ras pi) is running thermostat app

Attaching the logs from both (chip tool + chip tool Darwin) execution;

[TC-TSTAT3.12 -chiptool-log.txt](https://github.com/project-chip/connectedhomeip/files/8783109/TC-TSTAT3.12.-chiptool-log.txt)

[TC-TSTAT3.12_chiptooldarwin_log .txt](https://github.com/project-chip/connectedhomeip/files/8783110/TC-TSTAT3.12_chiptooldarwin_log.txt)

bzbarsky-apple, 2022-05-27:
GetRelayStatusLog is not part of Matter.  See the device library spec and https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/3392

closed 2022-05-27 by bzbarsky-apple

bzbarsky-apple, 2022-05-27:
And https://github.com/project-chip/connectedhomeip/pull/18880 to make this a bit more clear....