project-chip/connectedhomeip #18887: chip-tool: useless error messages in CASE resumption

bluebin14, 2022-05-27:
#### Problem
Lots of error messages in chip-tool when it is attempting to use CASE resumption:

```
[1653587684916] [96959:9058254] CHIP: [SC] Unable to load session resumption state during session deletion for node 0000000000BC5C01: ../../examples/chip-tool/config/PersistentStorage.cpp:108: CHIP Error 0x000000A0: Value not found in the persisted storag
[1653587684917] [96959:9058254] CHIP: [SC] Unable to delete session resumption state for node 0000000000BC5C01: ../../examples/chip-tool/config/PersistentStorage.cpp:140: CHIP Error 0x000000A0: Value not found in the persisted storage
```
This makes finding real errors in the output harder.

 #### Proposed Solution
Stop issuing error messages when trying to delete something that does not exist.

bzbarsky-apple, 2022-05-27:
This has nothing to do with chip-tool.  This is DefaultSessionResumptionStorage logging, and would happen on any device, not just chip-tool.

Probably the relevant logs in `DefaultSessionResumptionStorage::Delete` should check for `err != CHIP_NO_ERROR && err != CHIP_ERROR_PERSISTED_STORAGE_VALUE_NOT_FOUND`.

@bluebin14 Since you have an easy way to test, do you want to do the PR?

@kghost

bluebin14, 2022-05-27:
@bzbarsky-apple I can confirm that the following indeed fixes the issue:

```
diff --git a/src/protocols/secure_channel/DefaultSessionResumptionStorage.cpp b/src/protocols/secure_channel/DefaultSessionResumptionStorage.cpp
index f08c19101..ad2df93bb 100644
--- a/src/protocols/secure_channel/DefaultSessionResumptionStorage.cpp
+++ b/src/protocols/secure_channel/DefaultSessionResumptionStorage.cpp
@@ -87,7 +87,7 @@ CHIP_ERROR DefaultSessionResumptionStorage::Delete(const ScopedNodeId & node)
                          ChipLogValueX64(node.GetNodeId()), err.Format());
         }
     }
-    else
+    else if (err != CHIP_ERROR_PERSISTED_STORAGE_VALUE_NOT_FOUND)
     {
         ChipLogError(SecureChannel,
                      "Unable to load session resumption state during session deletion for node " ChipLogFormatX64
@@ -96,7 +96,7 @@ CHIP_ERROR DefaultSessionResumptionStorage::Delete(const ScopedNodeId & node)
     }
 
     err = DeleteState(node);
-    if (err != CHIP_NO_ERROR)
+    if (err != CHIP_NO_ERROR && err != CHIP_ERROR_PERSISTED_STORAGE_VALUE_NOT_FOUND)
     {
         ChipLogError(SecureChannel, "Unable to delete session resumption state for node " ChipLogFormatX64 ": %" CHIP_ERROR_FORMAT,
                      ChipLogValueX64(node.GetNodeId()), err.Format());

```

closed 2022-05-28 by bzbarsky-apple