project-chip/connectedhomeip #18889: TC-BC-2.1 Binding Cluster - binding attribute showing wrong number of entries

sethunk, 2022-05-27:
**Actual Behavior** - TH writes two binding entries in which the response is SUCCESS but during read, the response is only one entry (first entry) 
**Expected Behavior** - When TH writes two binding entries, the response should be SUCCESS and during read, the response should consist of two binding entries
**App used** - allclusters app
**Platform** - Chip-tool - RPI-4, 8GB RAM
                       DUT         - RPI-4, 8GB RAM
**Network** - Ble-wifi		   
**Commit id** - 425f1fb916ca154deeb75cd82cd5196778aba152
**Steps to reproduce** : 
Write binding entry : ./chip-tool binding write binding '[{"node" : 1  , "cluster" : "0x0006" , "endpoint" : 1 }] , [{ "group" : "0x0001" }]' 1 0
Read binding entry : ./chip-tool binding read binding 1 0

[DUT_Logs.txt](https://github.com/project-chip/connectedhomeip/files/8786498/DUT_Logs.txt)
[TH_Logs.txt](https://github.com/project-chip/connectedhomeip/files/8786499/TH_Logs.txt)

bzbarsky-apple, 2022-05-27:
>  Write binding entry : ./chip-tool binding write binding '[{"node" : 1 , "cluster" : "0x0006" , "endpoint" : 1 }] , [{ "group" : "0x0001" }]' 1 0

This command is wrong.  It's trying to write two lists (which I would expect to end up in an error in the parser; will look into why that does not happen).  But in any case, the first list is what gets written.  Fixing the command like so:

```
 Write binding entry : ./chip-tool binding write binding '[{"node" : 1 , "cluster" : "0x0006" , "endpoint" : 1 }, { "group" : "0x0001" }]' 1 0
```
gives the right behavior, as expected.

bzbarsky-apple, 2022-05-27:
@sethunk Did that command come from the verification spreadsheet?

closed 2022-05-27 by bzbarsky-apple