project-chip/connectedhomeip #18892: TC-GR-2.2 Groups Cluster - groups are not removed from GroupTable even after RemoveGroup command gets Success response

sethunk, 2022-05-27:
**Problem** - Even after TH removes the group using RemoveGroup command, that particular group exists in GroupTable

**Actual Behavior** - TH removes the group using RemoveGroup command and the response is Success, then TH reads GroupTable attribute in which the removed group exists.

**Expected Behavior** - After TH removes the group using RemoveGroup command then that group should not be present in the GroupTable attribute entries.

**App used** - allclusters app
**Platform** - Chip-tool - RPI-4, 8GB RAM
                       DUT         -  RPI-4, 8GB RAM
**Network** - Ble-wifi
**Commit** - 425f1fb916ca154deeb75cd82cd5196778aba152

**Steps to reproduce** - 
removegroup - ./chip-tool groups remove-group 0x0001 1 0
viewgroup      - ./chip-tool groups view-group 0x0001 1 0
grouptable     - ./chip-tool groupkeymanagement read group-table 1 0

[TH_Logs.txt](https://github.com/project-chip/connectedhomeip/files/8786810/TH_Logs.txt)
[DUT_Logs.txt](https://github.com/project-chip/connectedhomeip/files/8786811/DUT_Logs.txt)

woody-apple, 2022-05-27:
SDK Spec Review: Removing spec, as we believe we can make the SDK have a reasonable behavior here.

bzbarsky-apple, 2022-05-28:
The "spec" label means "SDK is not following the spec".  Which in this case we are not....

jepenven-silabs, 2022-05-30:
This is due to the fact that the GroupTable isn't really a GroupTable but more of a Group/Group Key binding table. So in this case, you're removing a Group and NOT a binding. If I remember correctly the spec doesn't enforce the fact that upon a group removal you need to remove all related bindings (it would make sense to do so). 

@rcasallas-silabs @turon

jepenven-silabs, 2022-05-30:
Already had this discussion with @turon here : https://github.com/CHIP-Specifications/chip-test-plans/pull/1006#discussion_r845121983

rcasallas-silabs, 2022-05-31:
The problem, I believe, is that Group cluster, and Group Key Management cluster have mutually exclusive requirements. According to section 1.3.7.4. (RemoveGroup Command): `The RemoveGroup command allows a client to request that the server removes the membership for the endpoint implementing the server, if any, in a particular group`.
This implies that the endpoint to group association must be removed, but point 3 adds:
`3. The server removes the requested entry from the Group Table and the status SHALL be SUCCESS.`

However, in the main spec, section 11.2.2.3. (GroupInfoMapStruct), each GroupInfo (entries of the Group Table) must be:
```
GroupId     group-id
Endpoints   list[endpoint-no]
GroupName   string
```
So, removing one endpoint from the group must NOT remove the entire entry of the Group Table, for this would also remove ALL endpoints from the group.

However, what should happen when the last remaining endpoint is removed from a GroupInfo? Should then the GroupInfo be automatically removed from the Group Table? Is there such requirement?

tcarmelveilleux, 2022-06-01:
This seems to be related to having multiple idea of "Group Table".

There is the GroupKeyManagement `GroupTable` attribute that relates the sum of the groups configured on all endpoints, by group ID. It's a read-only attribute.

There are the Group commands in the Groups cluster, which are per-endpoint and *per fabric*. The output of ViewGroup is <status, group ID, group Name> for a given group on a given endpoint.

There is also the "Logical concept of a group table" ("Group Table" as is appears in Groups cluster), which is where the prose is complete.

The Group Data Provider has:

```
    virtual bool HasEndpoint(FabricIndex fabric_index, GroupId group_id, EndpointId endpoint_id)          = 0;
    virtual CHIP_ERROR AddEndpoint(FabricIndex fabric_index, GroupId group_id, EndpointId endpoint_id)    = 0;
    virtual CHIP_ERROR RemoveEndpoint(FabricIndex fabric_index, GroupId group_id, EndpointId endpoint_id) = 0;
    virtual CHIP_ERROR RemoveEndpoint(FabricIndex fabric_index, EndpointId endpoint_id)    
```

This affects the mapping of a group ID for an endpoint.

`AddGroup` command handler does:

```
static EmberAfStatus GroupAdd(FabricIndex fabricIndex, EndpointId endpointId, GroupId groupId, const CharSpan & groupName)
{
    VerifyOrReturnError(IsFabricGroupId(groupId), EMBER_ZCL_STATUS_INVALID_VALUE);

    GroupDataProvider * provider = GetGroupDataProvider();
    VerifyOrReturnError(nullptr != provider, EMBER_ZCL_STATUS_NOT_FOUND);

    CHIP_ERROR err = provider->SetGroupInfo(fabricIndex, GroupDataProvider::GroupInfo(groupId, groupName));
    if (CHIP_NO_ERROR == err)
    {
        err = provider->AddEndpoint(fabricIndex, groupId, endpointId);
    }
    if (CHIP_NO_ERROR == err)
    {
        return EMBER_ZCL_STATUS_SUCCESS;
    }

    ChipLogDetail(Zcl, "ERR: Failed to add mapping (end:%d, group:0x%x), err:%" CHIP_ERROR_FORMAT, endpointId, groupId,
                  err.Format());
    return EMBER_ZCL_STATUS_INSUFFICIENT_SPACE;
}
```

--> Sets group info for a given group ID (which, for instance, renames across all endpoints, since a group has a single name across all endpoints

To me, the whole thing makes perfect sense, but it's clumsily expressed.

For a given fabric:

- You AddGroup per endpoint with Groups cluster, which calls both `provider->SetGroupInfo` and `provider->AddEndpoint`
- You RemoveGroup per endpoint with groups cluster. The GroupKeyManagements `Groups` attribute will show one less entry in the Endpoints list for a given group ID. If a group is no longer bound to any endpoint, it no longer appears in the GroupKeyManagement `Groups` list at all.

In terms of actions, in sequence, starting empty, and assuming no name support (omitting names), I would expect:

1. READ EP0/GroupKeyManagement.GroupTable == []
2. WRITE EP0/GroupKeyManagement.GroupKeyMap[0] = <Group 222, Keyset 11222>
3. INVOKE EP2/Groups.AddGroup(222)
4. READ EP0/GroupKeyManagement.GroupTable == [<GroupId: 222, Endpoints: [2]>]
5. INVOKE EP5/Groups.AddGroup(222)
6. READ EP0/GroupKeyManagement.GroupTable == [<GroupId: 222, Endpoints: [2, 5]>]
7. INVOKE EP2/Groups.RemoveGroup(222)
8. READ EP0/GroupKeyManagement.GroupTable == [<GroupId: 222, Endpoints: [5]>]
9. INVOKE EP5/Groups.RemoveGroup(222)
10. READ EP0/GroupKeyManagement.GroupTable == [] 

Note how at after step 9, we no longer have the GroupId existing at all. 

If after Step 9, it was:

READ EP0/GroupKeyManagement.GroupTable == [<GroupId: 222, Endpoints: []>]

Then we would be in a situation where the `GroupTable` attribute can always have groups that are basically non-responsive, and help nobody. GroupTable is supposed to reflect the groups added via AddGroup and removed via RemoveGroup

rcasallas-silabs, 2022-06-01:
Agreed. Now we should put that on the spec, so the current behavior can be considered a bug, and be fixed.

tcarmelveilleux, 2022-06-01:
Need to implement the result of:

- https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5341
- https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5342