project-chip/connectedhomeip #18901: scan-networks returns 0 entry

cabin15, 2022-05-27:
#### Problem
scan-networks command returns 0 entry on ScanNetworkResponse.

Expected result (Suppose to display more than one access point information.
[1653665629.998533][284849:284854] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0001
[1653665629.998596][284849:284854] CHIP:TOO:   ScanNetworksResponse: {
[1653665629.998619][284849:284854] CHIP:TOO:     networkingStatus: 0
[1653665629.998649][284849:284854] CHIP:TOO:     wiFiScanResults: 1 entries
[1653665629.998703][284849:284854] CHIP:TOO:       [1]: {
[1653665629.998737][284849:284854] CHIP:TOO:         Security: 8
[1653665629.998773][284849:284854] CHIP:TOO:         Ssid: 636869705F7373685F74657374
[1653665629.998806][284849:284854] CHIP:TOO:         Bssid: 909F3365A878
[1653665629.998837][284849:284854] CHIP:TOO:         Channel: 149
[1653665629.998869][284849:284854] CHIP:TOO:         WiFiBand: 2
[1653665629.998897][284849:284854] CHIP:TOO:         Rssi: -39
[1653665629.998923][284849:284854] CHIP:TOO:        }
[1653665629.998954][284849:284854] CHIP:TOO:    }
[1653665629.999018][284849:284854] CHIP:DMG: ICR moving to [AwaitingDe]

Actual result
[1653665337.325238][284577:284582] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0001
[1653665337.325302][284577:284582] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0001
[1653665337.325389][284577:284582] CHIP:TOO:   ScanNetworksResponse: {
[1653665337.325444][284577:284582] CHIP:TOO:     networkingStatus: 0
[1653665337.325480][284577:284582] CHIP:TOO:     wiFiScanResults: 0 entries
[1653665337.325513][284577:284582] CHIP:TOO:    }
[1653665337.325571][284577:284582] CHIP:DMG: ICR moving to [AwaitingDe]

[all-clusters-app-scan-networks-0-entry.txt](https://github.com/project-chip/connectedhomeip/files/8787755/all-clusters-app-scan-networks-0-entry.txt)
[chip-tool-scan-networks-0-entry.txt](https://github.com/project-chip/connectedhomeip/files/8787756/chip-tool-scan-networks-0-entry.txt)

bzbarsky-apple, 2022-05-28:
That is very strange.  The platform code shows a bunch of networks found:
```
...
[1653665337.001872][211435:211438] CHIP:DL: Network Found: NETGEAR19 (08:02:8e:ff:c7:2b) Signal:-31
[1653665337.005350][211435:211438] CHIP:DL: Network Found: moon (38:d5:47:63:57:b0) Signal:-38
[1653665337.009942][211435:211438] CHIP:DL: Network Found: shlee-NETGEAR65 (a0:04:60:e7:64:c0) Signal:-59
...
```
And the response command is sent after that point, but the response definitely shows 0 entries..

@yufengwangca any idea what could be going on here?

woody-apple, 2022-06-15:
Spec Issue Review: Tagging @franck-apple to follow up with Suma or Sankar to verify there's a test case that covers this, as well as the steps, and why this isn't a cert blocker if this is still failing.

woody-apple, 2022-06-21:
Cert Blocker Review: Marking as cert blocker, we think something is broken on the linux platform. Tagging @andy31415 @franck-apple myself @woody-apple to find an owner.

cjandhyala, 2022-06-28:
@cabin15 Did you provision the device on IP transport , if so, then the response returns 0 entries

andy31415, 2022-07-05:
@cabin15 - please also provide the command lines that you used. The report says `scan-networks command returns 0 entry` but it does not describe how scan-networks was actually called. Vivien found that using `Ssid null` can cause issues, but I cannot see if that was your use case or not.

https://github.com/project-chip/connectedhomeip/blob/master/docs/BUG_REPORT.md#reporting-bugs

andy31415, 2022-07-05:
From the log I see that the reported command is:

```
./chip-tool networkcommissioning scan-networks 1 0 --Ssid null
```

ssid null is indeed very suspect

cabin15, 2022-07-07:
@andy31415 
Using latest masterwhere #20312 is merged, I can see a multiple entries of ScanNetworkResponse upon below command.

./chip-tool networkcommissioning scan-networks 1 0 --Ssid null

Is this still a valid command to perform the test?
Thank you for the fix.

andy31415, 2022-07-09:
Marking clsed.

I am unsure if ssid null was supposed to be valid ... but I see it is valid now, so probably can use it. I would have preferred a better argument than null (like not specifying ssid to remove all filtering) however that can be updated separately.

closed 2022-07-09 by andy31415