project-chip/connectedhomeip #18928: [pre-sve] TC-I-1.1 AcceptedCommandList doesnot show command with id 0x01 and GeneratedCommandList doesnt shows command with id 0

PSONALl, 2022-05-30:
#### Problem
 * expected behavior vs actual behavior
      -  TH reads the AcceptedCommandList attribute from the DUT and This list should include the mandatory commands (values 0x00, and 0x01) but the list has only values 0, 64
      - TH reads the GeneratedCommandList attribute from the DUT and This list should include the mandatory response command (value 0x00) but the list is null

 * system configuration
  * system configuration :
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
    
 * logs
      - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8795732/esp32.txt)
      - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8795734/th.txt)

PSONALl, 2022-05-30:
@cjandhyala this is a  cert blocker

cjandhyala, 2022-05-30:
@PSONALl Based on the resolution #18929 , the test plan needs to be updated per the latest spec changes and the issue is not on SDK

cjandhyala, 2022-05-30:
@PSONALl Based on the resolution https://github.com/project-chip/connectedhomeip/issues/18929 , the test plan needs to be updated per the latest spec changes and the issue on the SDK side is that per spec reference app should not have returned the AcceptedCommandList with 64. Since this is not optional command, I labeling this as spec compliance rather than a cert blocker 

[image]

cjandhyala, 2022-05-31:
@PSONALl #18929 discussion, we need SDK to implement QRY feature, until then GeneratedCommandList shows null value.

bzbarsky-apple, 2022-06-02:
QRY is not part of Matter, so the SDK behavior here is correct.

closed 2022-06-02 by bzbarsky-apple