project-chip/connectedhomeip #18929: [pre-sve] TC-i-2.2 1. step 3- Identify-query command is not supported by chip-tool  2. step - 6 a after writing value on identify when we read it it returns 0 always

PSONALl, 2022-05-30:
#### Problem

 * expected behavior vs actual behavior
        - TH should send IdentifyQuery command to DUT but the command is not supported in chip-tool
        - Reading value of Identify time attribute immediately after write must return value approximately equal to value written but it always returns 0

 * system configuration
      -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
    
 * logs
   - esp32: [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8795923/esp32.txt)
   - chip-tool: [th.txt](https://github.com/project-chip/connectedhomeip/files/8795934/th.txt)

PSONALl, 2022-05-30:
@cjandhyala this is a  cert blocker

andy31415, 2022-05-30:
Need to determine if this happens on all platforms or only some. I believe we mostly validated with RPI and nrf, so I expect other platforms to maybe have additional issues.

I believe we should track platform-specific cert blockers in some way to be clearly identifiable.

cjandhyala, 2022-05-30:
@andy31415 @PSONALl , this is not an SDK issue, the spec was updated last week, Ref https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5240, There is a ticket https://github.com/CHIP-Specifications/chip-test-plans/issues/1389 on the test plans team to update the test plan.

cjandhyala, 2022-05-30:
closing per above comments

cjandhyala, 2022-05-31:
SDK to implement #18973

closed 2022-05-31 by bzbarsky-apple