project-chip/connectedhomeip #18931: Source bootstrap.sh error

TheWalkingDead1024, 2022-05-30:
When I try to source bootstrap.sh to build preparation for CHIP project (I use M1 Macbook Pro), I got an error like this(I am using branch master(the latest commit is f188fd9625f171a7d32a4b8e4a00d6abf60af891), python verson 3.9.6):

      In file included from third_party/boringssl-with-bazel/err_data.c:17:
      third_party/boringssl-with-bazel/src/include/openssl/base.h:187:9: warning: 'OPENSSL_VERSION_NUMBER' macro redefined [-Wmacro-redefined]
      #define OPENSSL_VERSION_NUMBER 0x1010107f
              ^
      /opt/homebrew/opt/openssl@3/include/openssl/opensslv.h:99:10: note: previous definition is here
      # define OPENSSL_VERSION_NUMBER          \
               ^
      In file included from third_party/boringssl-with-bazel/err_data.c:18:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/err.h:21:
      /opt/homebrew/opt/openssl@3/include/openssl/e_os2.h:183:11: warning: 'OPENSSL_EXPORT' macro redefined [-Wmacro-redefined]
      #  define OPENSSL_EXPORT extern
                ^
      third_party/boringssl-with-bazel/src/include/openssl/base.h:222:9: note: previous definition is here
      #define OPENSSL_EXPORT
              ^
      In file included from third_party/boringssl-with-bazel/err_data.c:18:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/err.h:28:
      /opt/homebrew/opt/openssl@3/include/openssl/types.h:62:13: error: typedef redefinition with different types ('int' vs 'struct asn1_null_st')
      typedef int ASN1_NULL;
                  ^
      third_party/boringssl-with-bazel/src/include/openssl/base.h:333:29: note: previous definition is here
      typedef struct asn1_null_st ASN1_NULL;
                                  ^
      In file included from third_party/boringssl-with-bazel/err_data.c:18:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/err.h:28:
      /opt/homebrew/opt/openssl@3/include/openssl/types.h:103:26: error: typedef redefinition with different types ('struct evp_md_st' vs 'struct env_md_st')
      typedef struct evp_md_st EVP_MD;
                               ^
      third_party/boringssl-with-bazel/src/include/openssl/base.h:403:26: note: previous definition is here
      typedef struct env_md_st EVP_MD;
                               ^
      In file included from third_party/boringssl-with-bazel/err_data.c:18:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/err.h:28:
      /opt/homebrew/opt/openssl@3/include/openssl/types.h:104:30: error: typedef redefinition with different types ('struct evp_md_ctx_st' vs 'struct env_md_ctx_st')
      typedef struct evp_md_ctx_st EVP_MD_CTX;
                                   ^
      third_party/boringssl-with-bazel/src/include/openssl/base.h:402:30: note: previous definition is here
      typedef struct env_md_ctx_st EVP_MD_CTX;
                                   ^
      In file included from third_party/boringssl-with-bazel/err_data.c:18:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/err.h:28:
      /opt/homebrew/opt/openssl@3/include/openssl/types.h:130:34: error: typedef redefinition with different types ('struct evp_Encode_Ctx_st' vs 'struct evp_encode_ctx_st')
      typedef struct evp_Encode_Ctx_st EVP_ENCODE_CTX;
                                       ^
      third_party/boringssl-with-bazel/src/include/openssl/base.h:407:34: note: previous definition is here
      typedef struct evp_encode_ctx_st EVP_ENCODE_CTX;
                                       ^
      In file included from third_party/boringssl-with-bazel/err_data.c:18:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/err.h:29:
      In file included from /opt/homebrew/opt/openssl@3/include/openssl/bio.h:30:
      /opt/homebrew/opt/openssl@3/include/openssl/crypto.h:308:3: error: typedef redefinition with different types ('struct crypto_threadid_st' vs 'int')
      } CRYPTO_THREADID;
        ^
      third_party/boringssl-with-bazel/src/include/openssl/base.h:329:13: note: previous definition is here
      typedef int CRYPTO_THREADID;
                  ^
      third_party/boringssl-with-bazel/err_data.c:33:1: error: static_assert failed due to requirement '13 == 12' "library value changed"
      OPENSSL_STATIC_ASSERT(ERR_LIB_ASN1 == 12, "library value changed");
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      third_party/boringssl-with-bazel/src/include/openssl/type_check.h:75:42: note: expanded from macro 'OPENSSL_STATIC_ASSERT'
      #define OPENSSL_STATIC_ASSERT(cond, msg) _Static_assert(cond, msg)
                                               ^              ~~~~
      third_party/boringssl-with-bazel/err_data.c:34:1: error: static_assert failed due to requirement '14 == 13' "library value changed"
      OPENSSL_STATIC_ASSERT(ERR_LIB_CONF == 13, "library value changed");
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      third_party/boringssl-with-bazel/src/include/openssl/type_check.h:75:42: note: expanded from macro 'OPENSSL_STATIC_ASSERT'
      #define OPENSSL_STATIC_ASSERT(cond, msg) _Static_assert(cond, msg)
                                               ^              ~~~~
[... 234 lines cut ...]
['ninja', '-C', '/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 287, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 768, in <module>
    sys.exit(main())
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 760, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 442, in setup
    result = step(spin)
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 582, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 302, in install
    install_packages(gn_target)
  File "/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 290, in install_packages
    raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/TheWalkingDead/Documents/GitHub/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

bzbarsky-apple, 2022-06-02:
@mspang any idea what's going on there?

benjaminnestler, 2022-08-24:
What about the first errors relating the multiple defines of OpenSSL (installed via brew) and boringssl-with-bazel?

closed 2022-10-20 by TheWalkingDead1024