project-chip/connectedhomeip #18938: [pre-sve] I-3.1 If value written on IdentifyTime attribute is invalid,  appropriate feedback is not provided to DUT

PSONALl, 2022-05-30:
#### Problem
 * expected behavior
     - TH receives the write command, checks whether the value is valid. If value invalid, provide appropriate feedback to DUT and flag test as FAIL
 * actual behavior
     - no any feedback is provided to DUT if the value is invalid

  * system configuration :
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
 * logs
       - esp32 [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8797449/esp32.txt)
       - chip-tool [th.txt](https://github.com/project-chip/connectedhomeip/files/8797461/th.txt)

PSONALl, 2022-05-30:
@cjandhyala please mark it as a cert blocker

cjandhyala, 2022-05-30:
@PSONALl This is not a SDK issue , chip-tool is filtering the invalid value. if you need to write an invalid value , you need to use write-by-id  , you will receive CONSTRAINT_ERROR back.

`sudo ./out/debug/chip-tool identify write-by-id 0 65536 60 0`

closing the issue

closed 2022-05-30 by cjandhyala