project-chip/connectedhomeip #18940: [TC-MC-1.10] attributeList value mismatch error

manjunath-grl, 2022-05-30:
Summary:
As per testplan of TC-MC-1.10 attribute value must be [0,1,2,3,4,5,6,7,65528,65529,65531,65532,65533] but got [0,1,2,3,5,6,7,65528,65529,65531,65532,65533] when executed value 4 (Application attribute) is missing.

Testplan screenshot:
 [image]
Spec screenshot:
[image]
Execution logs:
[Execution_log.txt](https://github.com/project-chip/connectedhomeip/files/8798119/Execution_log.txt)

mrjerryjohns, 2022-05-31:
On master as of today, (`eb3ac837a`), I'm seeing attribute 4 show up in the supported attribute list.

However, the list of attributes returned don't seem to match that in the YAML:

Values I read out:
0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 65528, 65529, 65531, 65532, 65533

Values in the YAML test:
0, 1, 2, 3, 4, 5, 6, 7, 65528, 65529, 65531, 65532, 65533

cjandhyala, 2022-06-01:
@manjunath-grl Can you provide the details of your test environment and which app are you using ?

manjunath-grl, 2022-06-01:
@cjandhyala we are testing this in Linux system and testing it against all clusters app

bzbarsky-apple, 2022-06-02:
This is a media cluster, so should probably be tested against tv-app....

manjunath-grl, 2022-06-02:
@cjandhyala @bzbarsky-apple we are using the commands used for CI
 For Build:
 ./scripts/run_in_build_env.sh
 "./scripts/build/build_examples.py
 --target linux-x64-chip-tool-no-ble-tsan-clang
 --target linux-x64-all-clusters-no-ble-tsan-clang
 --target linux-x64-lock-no-ble-tsan-clang
 --target linux-x64-ota-provider-no-ble-tsan-clang
 --target linux-x64-ota-requestor-no-ble-tsan-clang
 --target linux-x64-tv-app-no-ble-tsan-clang
 build
 "
We are testing this against TV-app

bzbarsky-apple, 2022-06-02:
@manjunath-grl CI runs test files starting with "TV_" against tv-app, file starting with "DL_" against door-lock-app, and other things against all-clusters-app.

manjunath-grl, 2022-06-15:
@bzbarsky-apple @franck-apple tried in TOT its passing
    SHA used: https://github.com/project-chip/connectedhomeip/commit/c1e1d018820c0fdd51f124f607b9207f31fae2ff
    Client app: Chip-tool
    Server app: Tv-app
    Execution logs: [TC_MC_1_10 ACL.zip](https://github.com/project-chip/connectedhomeip/files/8906445/TC_MC_1_10.ACL.zip)

closed 2022-06-15 by manjunath-grl