project-chip/connectedhomeip #18942: [pre-sve] TC-oo-2.3 on-with-timed-off command doesnot change values of On-Time Off-Wait-Time and OnOff attribute

PSONALl, 2022-05-30:
#### Problem

 * expected behavior
    - In step 10e after sending on-with-timed-off command i. The value of OnOff has to be TRUE. ii. The value of OnTime has to be approximately equal to 0x012c. iii. The value of OffWaitTime has to be 0x012c.
 * actual behavior: 
       - The value of OnOff is false, The value of OnTime and OffWaitTime is 0

 * system configuration
         -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 * logs
        - esp32 [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8798491/esp32.txt)
        - chip-tool [th.txt](https://github.com/project-chip/connectedhomeip/files/8798492/th.txt)

PSONALl, 2022-05-30:
@cjandhyala this is a cert blocker

cjandhyala, 2022-05-30:
@PSONALl The logs attached doesnt seem to belong to the scenario under test. Pls attach correct logs

mkleszcz-no, 2022-06-03:
I have similar problem, but in my case in step 11b and 12b, expected value of ONOff is false. got true

logs:
[TC_OO_2_3_nRF_DUT.log](https://github.com/project-chip/connectedhomeip/files/8833382/TC_OO_2_3_nRF_DUT.log)
[TC_OO_2_3_nRF.log](https://github.com/project-chip/connectedhomeip/files/8833385/TC_OO_2_3_nRF.log)

PSONALl, 2022-06-13:
> @PSONALl The logs attached doesnt seem to belong to the scenario under test. Pls attach correct logs

@cjandhyala I have attached the correct logs now, please check.

PSONALl, 2022-06-20:
This test case is pass and the issue is solved on week 2 SHA: 5d8599d195a93dfe5473f64e4f888c322e5df1d2 
So closing this issue

closed 2022-06-20 by PSONALl