project-chip/connectedhomeip #18943: [pre-sve] TC-oo-2.3 off-with-effect command doesnot change value of OffWaitTime attribute

PSONALl, 2022-05-30:
#### Problem

 * expected behavior
    - OffWaitTime must be changed to the approximate value sent in command off-with-effect 
 * actual behavior
    - OffWithEffect attribute's value is always 0

 * system configuration
      -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : 425f1fb916ca154deeb75cd82cd5196778aba152 (pre-sve SHA)
        
 * logs
        - esp32 [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8798491/esp32.txt)
        - chip-tool [th.txt](https://github.com/project-chip/connectedhomeip/files/8798492/th.txt)

PSONALl, 2022-05-30:
@cjandhyala this is a cert blocker

somu1710, 2022-05-31:
@PSONALl  could u pls provide the ON/OFF cluster logs where you getting the error.you attached the identify cluster logs,could u pls check this?

PSONALl, 2022-05-31:
> @PSONALl could u pls provide the ON/OFF cluster logs where you getting the error.you attached the identify cluster logs,could u pls check this?

@somu1710 Updated the logs, please check

bzbarsky-apple, 2022-06-02:
@jmartinez-silabs could you take a look, please?

cjandhyala, 2022-06-02:
@PSONALl , I heard from @somu1710 that you got new instructions on how to execute this TC. Pls let us know if this is still an issue that we need SDK to look at it?

closed 2022-06-02 by jmartinez-silabs