project-chip/connectedhomeip #18973: SDK to implement 'Query Feature' in Identify cluster

cjandhyala, 2022-05-31:
With Latest ToT a2c7af21976195bdae4e2667448d87b70d1651a7,  SDK to implement 'Query Feature' in Identify cluster. 

Expected behavior: FeatureMap bit is set to 1.  Reference apps should have all features implemented to verify mandatory commands/attributes during cert/Pre-SVE testing. 

Actual behavior: FeatureMap bit is set to 0. 

Platform tested: All-cluster-app on raspi, chip-tool on mac. 

How to verify 
 % sudo ./out/debug/chip-tool identify read feature-map 60 0

[1653958559958] [34919:9215613] CHIP: [EM] Removed CHIP MessageCounter:3411946 from RetransTable on exchange 1430i
[1653958559958] [34919:9215613] CHIP: [DMG] ReportDataMessage =
[1653958559958] [34919:9215613] CHIP: [DMG] {
[1653958559959] [34919:9215613] CHIP: [DMG] 	AttributeReportIBs =
[1653958559959] [34919:9215613] CHIP: [DMG] 	[
[1653958559959] [34919:9215613] CHIP: [DMG] 		AttributeReportIB =
[1653958559959] [34919:9215613] CHIP: [DMG] 		{
[1653958559959] [34919:9215613] CHIP: [DMG] 			AttributeDataIB =
[1653958559959] [34919:9215613] CHIP: [DMG] 			{
[1653958559959] [34919:9215613] CHIP: [DMG] 				DataVersion = 0xdfe209b3,
[1653958559959] [34919:9215613] CHIP: [DMG] 				AttributePathIB =
[1653958559959] [34919:9215613] CHIP: [DMG] 				{
[1653958559959] [34919:9215613] CHIP: [DMG] 					Endpoint = 0x0,
[1653958559959] [34919:9215613] CHIP: [DMG] 					Cluster = 0x3,
[1653958559959] [34919:9215613] CHIP: [DMG] 					Attribute = 0x0000_FFFC,
[1653958559959] [34919:9215613] CHIP: [DMG] 				}
[1653958559959] [34919:9215613] CHIP: [DMG] 					
[1653958559959] [34919:9215613] CHIP: [DMG] 					Data = 0, 
[1653958559959] [34919:9215613] CHIP: [DMG] 			},
[1653958559959] [34919:9215613] CHIP: [DMG] 			
[1653958559959] [34919:9215613] CHIP: [DMG] 		},
[1653958559959] [34919:9215613] CHIP: [DMG] 		
[1653958559959] [34919:9215613] CHIP: [DMG] 	],
[1653958559959] [34919:9215613] CHIP: [DMG] 	
[1653958559959] [34919:9215613] CHIP: [DMG] 	SuppressResponse = true, 
[1653958559959] [34919:9215613] CHIP: [DMG] 	InteractionModelRevision = 1
[1653958559959] [34919:9215613] CHIP: [DMG] }
[1653958559959] [34919:9215613] CHIP: [TOO] Endpoint: 0 Cluster: 0x0000_0003 Attribute 0x0000_FFFC DataVersion: 3756132787
[1653958559960] [34919:9215613] CHIP: [TOO]   FeatureMap: 0

andy31415, 2022-05-31:
Are Feature maps not optional by definition?

andy31415, 2022-05-31:
Looking at the spec:

Feature QRY is a feature that controls `IdentifyQuery` and `IdentifyQueryResponse`.

My expectation is that we either say `Feature = 1` and implement the IdentifyQuery or we set `Feature = 0`.

bzbarsky-apple, 2022-05-31:
As in, Matter devices are not allowed to implement this feature.  It's Zigbee-only.

closed 2022-05-31 by bzbarsky-apple

tehampson, 2022-05-31:
>Matter devices are not allowed to implement this feature

But it sounds like the all-cluster example all is enabling this cluster? Are you saying that we should disable it, or that this isn't a required cluster of matter and therefore shouldn't be a cert blocker?

bzbarsky-apple, 2022-06-01:
The _cluster_ is enabled and supported.  The QRY feature, and anything conditioned on it, is not.