project-chip/connectedhomeip #18976: [pre-sve] TC-MF-1.3 second commissioner pairing timeout

pankore, 2022-05-31:
#### Problem
Test case ID: TC-MF-1.3
SHA: 425f1fb916ca154deeb75cd82cd5196778aba152
Platform: Ameba all-clusters-app
Chip-tool platform: Raspberry Pi
Pairing the second commissioner will timeout.

* Steps to reproduce:
Commission DUT using TH1
On TH1
`./chip-tool  pairing open-commissioning-window 1234 1 400 2000 3840`
On TH2
`./chip-tool pairing manualcode 1235 <manual pairing code> --commissioner-name beta`

[TC-MF-1.3-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8802014/TC-MF-1.3-chip-tool.txt)

sowmyassp, 2022-05-31:
With latest commit (4d84a4a978ffc867a62b9b15423a9b70d9cb993e) second commissioning is done successfully
**Chip-tool platform**: Raspberry Pi
**Steps to reproduce**:

- Commission DUT using TH1

1. On TH1
./chip-tool pairing open-commissioning-window 1234 1 400 2000 3840

1. On TH2
./chip-tool pairing manualcode 1235 <manual pairing code> --commissioner-name beta

**PFA logs**:
[DUT.txt](https://github.com/project-chip/connectedhomeip/files/8804004/DUT.txt)
[cntrl2.txt](https://github.com/project-chip/connectedhomeip/files/8804005/cntrl2.txt)
[cntrl1.txt](https://github.com/project-chip/connectedhomeip/files/8804006/cntrl1.txt)

cjandhyala, 2022-06-01:
@pankore Looking at the log, the timeout error from BLE can be ignored. from the log, it seems the controller (2nd admin) is able to provision the controlee over IP transport.  Can you try to read/write an attribute using the 2nd admin?

pankore, 2022-06-01:
@cjandhyala, yes it is working. Should I close this issue or will there be a fix for the BLE timeout error?

pankore, 2022-06-01:
@cjandhyala on TH2, I commissioned the DUT into fabric2 with nodeID=1235, but I could not control it using `./chip-tool basic read node-label 1235 0`. However, I can use TH2 to control it when I pass in nodeID of the DUT in fabric1 `./chip-tool basic read node-label 1234 0`.

cjandhyala, 2022-06-01:
@pankore Are you creating two fabrics with the same tool chip-tool instance on raspi. If yes, Can you give the command with commissioner's name 

`./chip-tool basic read node-label 1235 0 --commissioner-name beta`

pankore, 2022-06-01:
This solves the issue, thanks.