project-chip/connectedhomeip #18984: E (2843) chip[SVR]: Error when trying to join Group 257 of fabric index 1 : 10e

nykh2010, 2022-05-31:
when I use "./chip-tool groups add group 257 "grp" 1 0" to add esp32 to one group. it is fail when join the group. the error log is "E (2843) chip[SVR]: Error when trying to join Group 257 of fabric index 1 : 10e". 
I use the lighting-app example.
[multicastAddress.log](https://github.com/project-chip/connectedhomeip/files/8803315/multicastAddress.log)

mrjerryjohns, 2022-05-31:
`10e` is an `INET_ERROR_UNKNOWN_INTERFACE`.

mrjerryjohns, 2022-05-31:
The above error is being emitted from `Server::RejoinExistingMulticastGroups`, when trying to call `mTransports.MulticastGroupJoinLeave`. Someone versed in group comms should help take a look at this...

bzbarsky-apple, 2022-06-02:
@dhrishi @mkardous-silabs @turon

I expect that `UDPEndPointImplLwIP::IPv6JoinLeaveMulticastGroupImpl` or `UDPEndPointImplLwIP::IPv4JoinLeaveMulticastGroupImpl` is what's returning INET_ERROR_UNKNOWN_INTERFACE, when `FindNetifFromInterfaceId` returns null.  

Looks like `UDP::MulticastGroupJoinLeave` does:
```
    return mUDPEndPoint->LeaveMulticastGroup(mUDPEndPoint->GetBoundInterface(), address.GetIPAddress());
```
and I am wondering whether `mUDPEndPoint->GetBoundInterface()` is the "wildcard" interface id, in that the UDP endpoint is not bound to a specific interface at all.... I am willing to bet it is.

If that's what's going on, then that's also the thing that breaks group comms on Mac...

jepenven-silabs, 2022-06-02:
No work was done on our side for LwIP support of Multicast. There was a compatibility issue between the OpenThread Stack and the LwIP stack fro groups on top of code duplication hence we move aside from using LwIP and that context.

dtodor, 2022-06-13:
We are having the same issue. Is group messaging supposed to work on EFR32 and ESP32?

woody-apple, 2022-06-15:
Spec Issue Review: @rcasallas-silabs Are you able to take a look at this group issue?

jepenven-silabs, 2022-06-15:
This is platform specific to LwIP, issue is not present on Native Openthread platform nor on Zephyr platform. @Damian-Nordic can you confirm that it indeed works on Zephyr?

De-assigning Ricardo from this issue since its platform specific

jepenven-silabs, 2022-06-15:
More over if the device is using IPV4 this is expected behavior at that point since multicast doesn't support IPV4

Damian-Nordic, 2022-06-17:
@jepenven-silabs Yes, it definitely worked the last time I tried it on Zephyr platform and I believe we have automated tests for this functionality.

andy31415, 2022-07-06:
Looked a bit at 20136 - I believe our current code only uses mld6_joingroup_netif and never mld6_joingroup
For wildcard inet interface, we probably need to use mld6_joingroup

What I am unclear about is socket implementation ... there interface index 0 means `default interface` and I am unclear what default is. However if sockets work now, maybe we only need to fix lwip.

andy31415, 2022-07-06:
If group comms is also broken on mac, we may need group communication to use a separate endpoint for every interface (minmdns does this currently, but it also needed this to be able to send and receive unicast packets I believe).

woody-apple, 2022-07-13:
Cert Blocker Review: Closing this, given we believe this is resolved.