project-chip/connectedhomeip #18987: [TC-DL-2.8] Set User Command is giving Failure response

kowsisoundhar12, 2022-05-31:
Set User Command is giving Failure response while executing 

Testplan Screenshot:
[image]

Error Screenshot:
[image]

Yaml Scrrenshot:
[image]
[Execution-logs-DL-2-8.zip](https://github.com/project-chip/connectedhomeip/files/8804215/Execution-logs-DL-2-8.zip)

bzbarsky-apple, 2022-06-02:
@kowsisoundhar12 which server app is being tested here?

franck-apple, 2022-06-02:
removing "request cert blocker" per discussion with @manjunath-grl

bzbarsky-apple, 2022-06-18:
This was presumably being tested with all-clusters-app, since this is YAML, where that's not going to work.

closed 2022-06-18 by bzbarsky-apple