project-chip/connectedhomeip #18989: [TC-TFL-1.1]Time Format Localization Cluster--Feature-map value mismatch.

Rajashreekalmane, 2022-05-31:
**Problem**

As per test plan [TC-TFL-1.1] FeatureMap attribute showing value mismatch error when executed, DUT should
responds with feature map value 0 but currently DUT responding feature map value has 1.

**expected behavior:-** DUT should responds the feature map value as 0
**actual behavior:-** DUT is responding the feature map value as 1

**steps to reproduce:-** ./chip-tool timeformatlocalization read feature-map 1 0

**system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID:4d84a4a978ffc867a62b9b15423a9b70d9cb993e
Dut application: all-clusters-app

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/timeformatlocalization.adoc#tc-tfl-1-1-global-attributes-dut-as-client

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/LocalizationTimeFormat.adoc#features

**Spec Screenshot:-**
[image]

andy31415, 2022-05-31:
Feature 0 seems to be "CalendarFormat" and it seems to be controlling `ActiveCalendarType` and `SupportedCalendarTypes` attributes.

We should either:
   - have feature 0 disabled *or*
   - have feature 0 and support the 2 attributes it controls

@Rajashreekalmane could you confirm attributes available?

Looking at matter files, I generally see something like:

```
./examples/lighting-app/lighting-common/lighting-app.matter:  server cluster TimeFormatLocalization {
./examples/lighting-app/lighting-common/lighting-app.matter-    persist  attribute hourFormat;
./examples/lighting-app/lighting-common/lighting-app.matter-    persist  attribute activeCalendarType;
./examples/lighting-app/lighting-common/lighting-app.matter-    callback attribute supportedCalendarTypes;
./examples/lighting-app/lighting-common/lighting-app.matter-    ram      attribute featureMap;
./examples/lighting-app/lighting-common/lighting-app.matter-    ram      attribute clusterRevision default = 1;
./examples/lighting-app/lighting-common/lighting-app.matter-  }

./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter:  server cluster TimeFormatLocalization {
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    persist  attribute hourFormat;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    persist  attribute activeCalendarType;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    callback attribute supportedCalendarTypes;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    callback attribute generatedCommandList;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    callback attribute acceptedCommandList;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    callback attribute attributeList;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    ram      attribute featureMap;
./examples/all-clusters-app/all-clusters-common/all-clusters-app.matter-    ram      attribute clusterRevision default = 1;

./examples/lock-app/lock-common/lock-app.matter:  server cluster TimeFormatLocalization {
./examples/lock-app/lock-common/lock-app.matter-    persist  attribute hourFormat;
./examples/lock-app/lock-common/lock-app.matter-    persist  attribute activeCalendarType;
./examples/lock-app/lock-common/lock-app.matter-    callback attribute supportedCalendarTypes;
./examples/lock-app/lock-common/lock-app.matter-    ram      attribute featureMap;
./examples/lock-app/lock-common/lock-app.matter-    ram      attribute clusterRevision default = 1;
./examples/lock-app/lock-common/lock-app.matter-  }

```

Generally it seems that supportedCalendarTypes and activeCalendarType are supported, so featureMap being 1 is correct (although unclear why we do not default it here...).

Also notice that all-clusters has supported/generated/accepted commands but others like lock and lighting do not have it. I wonder if that is a cert issue.

mrjerryjohns, 2022-05-31:
So, on master, the feature map defaults to 0 now, even though on Linux/Darwin, it seems to read out valid values for active/supported calendar types on the all-clusters-app:

```
{
│   0: {
│   │   <class 'chip.clusters.Objects.TimeFormatLocalization'>: {
│   │   │   <class 'chip.clusters.Attribute.DataVersion'>: 162248214,
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.HourFormat'>: <HourFormat.k12hr: 0>,
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.ActiveCalendarType'>: <CalendarType.kBuddhist: 0>,
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.SupportedCalendarTypes'>: [
│   │   │   │   <CalendarType.kBuddhist: 0>,
│   │   │   │   <CalendarType.kChinese: 1>,
│   │   │   │   <CalendarType.kCoptic: 2>,
│   │   │   │   <CalendarType.kEthiopian: 3>,
│   │   │   │   <CalendarType.kGregorian: 4>,
│   │   │   │   <CalendarType.kHebrew: 5>,
│   │   │   │   <CalendarType.kIndian: 6>,
│   │   │   │   <CalendarType.kJapanese: 8>,
│   │   │   │   <CalendarType.kKorean: 9>,
│   │   │   │   <CalendarType.kPersian: 10>,
│   │   │   │   <CalendarType.kTaiwanese: 11>,
│   │   │   │   <CalendarType.kIslamic: 7>
│   │   │   ],
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.FeatureMap'>: 0,
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.ClusterRevision'>: 1,
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.GeneratedCommandList'>: [],
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.AcceptedCommandList'>: [],
│   │   │   <class 'chip.clusters.Objects.TimeFormatLocalization.Attributes.AttributeList'>: [
│   │   │   │   0,
│   │   │   │   1,
│   │   │   │   2,
│   │   │   │   65528,
│   │   │   │   65529,
│   │   │   │   65531,
│   │   │   │   65532,
│   │   │   │   65533
│   │   │   ]
│   │   }
│   }
}
```

Seems like the example apps _should_ be setting that feature map value to 1?

Rajashreekalmane, 2022-06-01:
@andy31415 DUT responding feature-map value has  0 and Currently CalendarFormat feature supporting 2 attributes those are ActiveCalendarType & SupportedCalendarTypes.

andy31415, 2022-06-01:
Cert blocker review:
   - feature map SHOULD be 1 because feature is supported (looks like it is 0 right now)
   - does not break the spec if feature map is 0 but attributes are available anyway

cjandhyala, 2023-06-27:
@andy31415 @mrjerryjohns  Could we fix the example all-cluster-app to return the correct featureMap value. Right now, the example app is returning 0 but the attribute list is returning other attributes ActiveCalendarType and SupportedCalendarTypes which should be available when the CALFMT feature is enabled.

Ashwinigrl, 2023-08-31:
Issue fixed in commit 2dc43d705ae41b559095b8cd80f143f5a0d02a8b.

Ashwinigrl, 2023-08-31:
Issue fixed in commit https://github.com/project-chip/connectedhomeip/commit/2dc43d705ae41b559095b8cd80f143f5a0d02a8b.
Hence Closing this issue

closed 2023-08-31 by Ashwinigrl