project-chip/connectedhomeip #18993: [TC-DGGEN-2.1] & [TC-DGGEN-3.1] -General Diagnostics-chip-tool needs to implement mandatory TestEventTriggersEnabled attribute.

Rajashreekalmane, 2022-05-31:
**Problem**

[TC-DGGEN-2.1]-General Diagnostics-chip-tool needs to implement mandatory TestEventTriggersEnabled attribute.

**PFA screenshot below:-**
[image]

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/general_diagnostics.adoc#tc-dggen-2-1-attributes-with-server-as-dut

**Spec reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/DiagnosticsGeneral.adoc

mrjerryjohns, 2022-05-31:
The commit reference doesn't contain support for this, which was recently added (see https://github.com/project-chip/connectedhomeip/pull/18852). Please re-test.

Rajashreekalmane, 2022-06-08:
@mrjerryjohns Still mandatory TestEventTriggersEnabled attribute is not implemented in chip-tool.

closed 2022-06-09 by Rajashreekalmane