project-chip/connectedhomeip #19000: Write attributes paths are not validated unlike read

fessehaeve, 2022-05-31:
#### Problem

`./chip-tool any read-by-id 0x11110001 0x1111f001 1 1 ` fails with INVALID_ACTION but 
`./chip-tool any write-by-id 0x11110001 0x1111f001 "abc" 1 1` succeeds. 

 #### Proposed Solution

Do we need to have this check?

mrjerryjohns, 2022-05-31:
On master as of today, the write does not actually succeed - it returns back an `UNSUPPORTED_CLUSTER` error.

This however isn't presumably the right error. The `WriteHandler` logic today does not actually validate that the Attribute and Cluster IDs are indeed well formed. This should be added.

CamWms, 2022-06-29:
Working on PR now. Should the action abort, or generate a INVALID_PATH?

CamWms, 2022-07-14:
I have a PR #5425 to check if paths are well-formed, but IDs should not be checked for in-range by the SDK code except if the range is within the MS range. The valid range is to direct new ID allocation, not for operational checks. We may expand or reduce the range in the future. If the SDK does not recognize an ID (in or outside the current range), then that is enough.

closed 2023-10-16 by bzbarsky-apple