project-chip/connectedhomeip #19036: [TC-LVL-7.1]--chip-tool needs to implement mandatory MoveToClosestFrequency command.

Rajashreekalmane, 2022-06-01:
#### Problem
chip-tool needs to implement mandatory MoveToClosestFrequency command.
 
**System configuration:-**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID:4d84a4a978ffc867a62b9b15423a9b70d9cb993e
Dut application: all-clusters-app

**PFA screenshot below:-**
[image]

**Test-plan reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/levelcontrol.adoc#tc-lvl-7-1-verification-of-movetoclosestfrequency-command-dut-as-server

**Spec reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/LevelControl.adoc#ref_sec_3_10_3_2

bzbarsky-apple, 2022-06-01:
This is not a chip-tool issue per se; the command is missing from the XML.

It's also not mandatory, but we clearly do need to add it.

bzbarsky-apple, 2022-06-08:
> is still not able to see the "MoveToClosestFrequency" in the Chip-tool command options

Using what SHA?  Using what steps?  @raju-apple @sankarselvam @Rajashreekalmane please actually provide relevant information when reporting problems; otherwise a lot of time gets wasted.

Concretely, on 5362125e31b8bdd7b2dddb2da381eb7bffe80f73 I see
```
% ./out/debug/standalone/chip-tool levelcontrol |& grep move-to-closest
  | * move-to-closest-frequency        
```
as expected.

closed 2022-06-08 by bzbarsky-apple

sankarselvam, 2022-06-08:
@Rajashreekalmane Can you please share the SHA you used?

Rajashreekalmane, 2022-06-09:
When it was tried in another setup, it worked fine.