project-chip/connectedhomeip #19039: [pre-sve] TC-LO-2.1 SupportedLocales and ActiveLocale issues

pankore, 2022-06-01:
#### Problem
Test case ID: TC-LO-2.1
SHA: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152
Platform: Ameba all-clusters-app
Chip-tool platform: Raspberry Pi

Ameba is using the DeviceInfoProviderImpl in examples/providers

- `./chip-tool localizationconfiguration read supported-locales 1234 0` Reading supported locales returns UNSUPPORTED_ATTRIBUTE, expected to be success.
-  `./chip-tool localizationconfiguration read active-locale 1234 0` Reading active locale returns Wrong TLV type, expected to be success
- `./chip-tool localizationconfiguration write active-locale fr-FR 1234 0` Writing active locale returns UNSUPPORTED_WRITE, expected to be CONSTRAINT_ERROR or success

[TC-LO-2.1-chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8812520/TC-LO-2.1-chip-tool.txt)

pankore, 2022-06-02:
Is this a platform specific issue? Seems like the DeviceInfoProviderImpl in examples/providers that ameba is using do support these read/write attributes.

pankore, 2022-06-03:
I have checked the ZAP file, supported-locales and active-locale attributes are enabled.

andy31415, 2022-06-03:
They are enabled as callbacks. You need to add support for DeviceInfoProviderImpl.

I believe for ameba, because a separate build script is used, those files could not be included in the app main. Your app main has to call the set provider api. See examples in:

https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/nrfconnect/main/AppTask.cpp#L155

or 

https://github.com/project-chip/connectedhomeip/blob/master/examples/lock-app/efr32/src/main.cpp#L49

Generally code should look like:

```
gExampleDeviceInfoProvider.SetStorageDelegate(&Server::GetInstance().GetPersistentStorage());
chip::DeviceLayer::SetDeviceInfoProvider(&gExampleDeviceInfoProvider);
```

andy31415, 2022-06-03:
This is platform specific: every platform example should call provides in main. We have implemented some examples (notably linux/mac, nrf and some efr) however we have not updated every example since build systems vary. Ameba build system is the most different one.

pankore, 2022-06-06:
Hi @andy31415, apologies I did not explain it clearly. In my local branch, I have included the `DeviceInfoProviderImpl` in Ameba's build and have called `SetDeviceInfoProvider` in main. I have not PR-ed this change.
In fact, reading supported calendar types is successful, which proves that the `DeviceInfoProviderImpl` is set. I am not sure why reading/writing locales fails.

pankore, 2022-06-09:
@woody-apple can you please re-review this again.
When TH sends `./chip-tool localizationconfiguration read supported-locales 1234 0`, DUT received a read request for cluster 0x2B, attributeID 2. However, supported-locales is attributeID 1.
Similarly for `./chip-tool localizationconfiguration read active-locale 1234 0`, DUT received read request for cluster 0x2B, attributeID 1. Active-locale should be attributeID 0.
See below log for reference.

[TC-LO-2.1-DUT.txt](https://github.com/project-chip/connectedhomeip/files/8867346/TC-LO-2.1-DUT.txt)

andy31415, 2022-06-09:
Does this happen on other platforms? 
If this happens on some other platforms (I believe CSG tests nrf and rpi at least) we should validate and then it could be moved to general issue.

Regarding the spec I agree we have:
   - LocalizationConfiguration cluster id is `0x002B`
   - ActiveLocale is a mandatory string attribute with id `0x0000`
   - SupportedLocales is a mandatory list of strings with id `0x0001`
   
If chip tool is asking for some different ids, it sounds like some sort of chip-tool bug (likely off-by-one). Would need confirmation this is the case on more than one platform (should be ... since chip-tool runs on linux and is common)

franck-apple, 2022-06-09:
@pankore please re-test on different platforms and update this issue with your findings. Thanks.

andy31415, 2022-06-09:
Ram the following test script on nrf-nrf52840dk-light on master

```sh
#!/usr/bin/env bash

set -ex

function finish {
  echo "STATUS: $?"
}

trap finish EXIT

export NODEID=$RANDOM

export THREAD_CRED=hex:<THREAD_CRED_HERE>
export CHIP_TOOL="$HOME/devel/connectedhomeip/out/linux-x64-chip-tool/chip-tool"

log_file="run_log_$(date +'%Y-%m-%d_%H-%M-%S').txt"
rm -f /tmp/chip_*

echo "NODE ID USED: $NODEID" | tee $log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL pairing ble-thread $NODEID $THREAD_CRED 20202021 3840 | tee -a $log_file
$CHIP_TOOL pairing ble-thread $NODEID $THREAD_CRED 20202021 3840 >>$log_file 2>&1

echo "=============================================================" >>$log_file
echo $CHIP_TOOL localizationconfiguration read active-locale $NODEID 0 | tee -a $log_file
$CHIP_TOOL localizationconfiguration read active-locale $NODEID 0  2>&1 | tee -a $log_file | grep 'ReportDataMessage =' -A 25

echo "=============================================================" >>$log_file
echo $CHIP_TOOL localizationconfiguration read supported-locales $NODEID 0 | tee -a $log_file
$CHIP_TOOL localizationconfiguration read supported-locales $NODEID 0  2>&1 | tee -a $log_file | grep 'ReportDataMessage =' -A 25
```

Logs contained

```
2190 [1654791355.680311][2958009:2958016] CHIP:TOO: Sending ReadAttribute to:
2191 [1654791355.680313][2958009:2958016] CHIP:TOO:  cluster 0x0000_002B, attribute: 0x0000_0000, endpoint 0
...
[1654791357.167104][2958042:2958049] CHIP:TOO: Sending ReadAttribute to:
[1654791357.167106][2958042:2958049] CHIP:TOO:  cluster 0x0000_002B, attribute: 0x0000_0001, endpoint 0
```

so I believe attribute ids seem spec-compilant

andy31415, 2022-06-09:
Data also seems available: logs say:

```
2903 [1654791357.343703][2958042:2958049] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_002B Attribute 0x0000_0001 DataVersion: 3713446426
2904 [1654791357.343728][2958042:2958049] CHIP:TOO:   SupportedLocales: 8 entries
2905 [1654791357.343734][2958042:2958049] CHIP:TOO:     [1]: en-US
2906 [1654791357.343737][2958042:2958049] CHIP:TOO:     [2]: de-DE
2907 [1654791357.343739][2958042:2958049] CHIP:TOO:     [3]: fr-FR
2908 [1654791357.343742][2958042:2958049] CHIP:TOO:     [4]: en-GB
2909 [1654791357.343744][2958042:2958049] CHIP:TOO:     [5]: es-ES
2910 [1654791357.343747][2958042:2958049] CHIP:TOO:     [6]: zh-CN
2911 [1654791357.343749][2958042:2958049] CHIP:TOO:     [7]: it-IT
2912 [1654791357.343752][2958042:2958049] CHIP:TOO:     [8]: ja-JP
```

pankore, 2022-06-10:
Tested on ToT's chip-tool with the same DUT firmware, got the same result as @andy31415 so it might be chip-tool's issue. Closing this issue as this is solved in ToT

closed 2022-06-10 by pankore