project-chip/connectedhomeip #19049: [pre-sve] TC-MF-1.10 BCM mode advertised as CM=0

jimlyall-q, 2022-06-01:
#### Problem
 * BCM opened and avahi-browse shows CM=1
 * BCM opened and avahi-browse shows CM=0 which shouldn't happen according to 4.3.1.3 "A Commissionee that is not in
commissioning mode (CM=0) SHALL NOT publish this subtype"
 * TC-MF-1.10
 * qpg6105 + x64 controller

```
txt = ["PI=" "PH=36" "CM=0" "D=3840" "T=0" "SAI=300" "SII=5000" "VP=65521+32773"]
= enp0s3 IPv6 6FCB71DD481A6B86                              _matterc._udp        local
hostname = [223BAECD839A1E85.local]
address = [fd11:22::b085:8ca5:8047:abee]
port = [5540]
```

 #### Proposed Solution

No fix

tcarmelveilleux, 2022-06-02:
How was BCM window opened?

bzbarsky-apple, 2022-06-02:
@jimlyall-q What exactly was being browsed for?

When CM=0, what should happen is:

* There is a TXT entry with key "CM" and value "0".  That's what I see in the `txt =` bits shown above, and that part is correct.
* The service name is not registered for the `_CM._matterc._udp` type+subtype (that's the "SHALL NOT publish this subtype" bit from the spec).  Was the browse being done for that subtype, or just for `_matterc._udp` across all subtypes?

bzbarsky-apple, 2022-06-02:
Also, above where it says "BCM opened and avahi-browse shows CM=0" is that supposed to be "BCM closed and ..."?  Because having CM=0 while a commissioning window is open would be _very_ unexpected.

jimlyall-q, 2022-06-03:
[TC_MF_1_10.1654071053.log](https://github.com/project-chip/connectedhomeip/files/8829854/TC_MF_1_10.1654071053.log)
Steps 2 open commissioning window - `chip-tool administratorcommissioning open-basic-commissioning-window 500 1 0 --timedInteractionTimeoutMs 1000`
Step 3 avahi-browse - `avahi-browse -rt _matterc._udp`

bzbarsky-apple, 2022-06-03:
>  Step 3 avahi-browse - avahi-browse -rt _matterc._udp

Thank you.

So if the server has extended discovery enabled, after initial commissioning it will advertise itself with CM=0.  You can see this in the log after network enable when we start getting advertisements like:
```
[1654071097.354913][3439:3444] CHIP:DIS: MINMDNS: received SRV record for 6FCB71DD481A6B86._matterc._udp.local
[1654071097.357259][3439:3444] CHIP:DIS: MINMDNS:     Commission SRV for instance 6FCB71DD481A6B86: 223BAECD839A1E85.local
...
[1654071097.357278][3439:3444] CHIP:DIS: MINMDNS: received PTR record for _services._dns-sd._udp.local
...
[1654071097.357290][3439:3444] CHIP:DIS: MINMDNS: received TXT record for 6FCB71DD481A6B86._matterc._udp.local
[1654071097.357299][3439:3444] CHIP:DIS: MINMDNS: received PTR record for _services._dns-sd._udp.local
[1654071097.357328][3439:3444] CHIP:DIS: MINMDNS: received PTR record for _L3840._sub._matterc._udp.local
[1654071097.357332][3439:3444] CHIP:DIS: MINMDNS: received PTR record for _S15._sub._matterc._udp.local
[1654071097.357337][3439:3444] CHIP:DIS: MINMDNS: received PTR record for _V65521._sub._matterc._udp.local
[1654071097.357341][3439:3444] CHIP:DIS: MINMDNS: received PTR record for _matterc._udp.local
```
The logging here does not show the contents of the TXT record, but I would assume it's got "CM=1" in it.  Note the lack of a "_CM.sub._matterc._udp.local" PTR record, because we are not in commissioning mode at that point.

When a commissioning window is then opened it will generate a new instance name and advertise with CM=1 (since it's a basic commissioning window).

The avahi-browse then shows the old records (because they get cached until TTL expires), so you see things like:
```
= docker0 IPv6 6FCB71DD481A6B86                              _matterc._udp        local
hostname = [223BAECD839A1E85.local]
address = [fd11:22::b085:8ca5:8047:abee]
port = [5540]
txt = ["PI=" "PH=36" "CM=0" "D=3840" "T=0" "SAI=300" "SII=5000" "VP=65521+32773"]
```
which has the same instance name as the log from minmdns above, and definitely has CM=0.

You also have the new records:
```
= docker0 IPv6 F24A94727E3F1C9A                              _matterc._udp        local
hostname = [223BAECD839A1E85.local]
address = [fd11:22::b085:8ca5:8047:abee]
port = [5540]
txt = ["PI=" "PH=36" "CM=1" "D=3840" "T=0" "SAI=300" "SII=5000" "VP=65521+32773"]
```
where the hostname is the same (since that's based on the MAC address), but the instance name is now `F24A94727E3F1C9A` instead of `6FCB71DD481A6B86`.  These are the records for the commmissionable advertisement, and they have CM=1.

Doing `avahi-browse -rt _CM._sub._matterc._udp` should work to only browse the things that have the _CM PTR record bits (i.e. only the things that are currently commissionable), and none of those should have CM=0 in the TXT records.

bzbarsky-apple, 2022-06-03:
Oh, and in step 4, when we are doing on-network discovery you can see:
```
[1654071102.239650][3455:3460] CHIP:DIS: MINMDNS: received SRV record for F24A94727E3F1C9A._matterc._udp.local
[1654071102.239743][3455:3460] CHIP:DIS: MINMDNS:     Commission SRV for instance F24A94727E3F1C9A: 223BAECD839A1E85.local
[1654071102.239770][3455:3460] CHIP:DIS: MINMDNS: received TXT record for F24A94727E3F1C9A._matterc._udp.local
[1654071102.239801][3455:3460] CHIP:DIS: MINMDNS: received PTR record for _CM._sub._matterc._udp.local
[1654071102.239827][3455:3460] CHIP:DIS: MINMDNS: received PTR record for _L3840._sub._matterc._udp.local
[1654071102.239850][3455:3460] CHIP:DIS: MINMDNS: received PTR record for _S15._sub._matterc._udp.local
[1654071102.239874][3455:3460] CHIP:DIS: MINMDNS: received PTR record for _V65521._sub._matterc._udp.local
[1654071102.239896][3455:3460] CHIP:DIS: MINMDNS: received PTR record for _matterc._udp.local
```
which is the advertising for the new instance name, and shows the _CM PTR record here, as expected.

jimlyall-q, 2022-06-07:
Thanks for spec help

closed 2022-06-07 by jimlyall-q