project-chip/connectedhomeip #19053: [Pre-SVE] test IDM 2.2 steps 5 and 8

HubertHugo, 2022-06-01:
Platform: nRF52840
all-cluster-app
After steps 5 and 8 (5th step requires setting timeout or it throws timeout before this error) there is 
[1654088492.058436][21409:21409] CHIP:TOO: Run command failure: IM Error 0x00000611: Cluster-specific error: 0x11
step 5 command:
./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF --timeout 40
step 8 command:
./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 1

cjandhyala, 2022-06-02:
@HubertHugo Could you attach the logs and provide bit more details the commands you sent prior to step5 /8

bzbarsky-apple, 2022-06-02:
TestCluster has an attribute that always fails reads, for testing purposes.  The above wildcard reads are reading everything including that attribute, so are getting that error logged.

closed 2022-06-02 by bzbarsky-apple

HubertHugo, 2022-06-02:
@cjandhyala if the problem is because of this always failing attribute then can I consider the test passed?
Here you have logs from 8th step.
[2.2.8_chip-tool.log](https://github.com/project-chip/connectedhomeip/files/8821928/2.2.8_chip-tool.log)