project-chip/connectedhomeip #19062: Actions Cluster - SDK needs to implement the Mandatory commands

sowmyassp, 2022-06-01:
**Problem**:
While Executing the behavior of the non-global commands of the Actions cluster, mandatory attribute throws error code as (unsupported command).

**Commit used** : 4d84a4a978ffc867a62b9b15423a9b70d9cb993e			

 **Expected behavior** :  Pass with status(0x0)

 **Actual behavior** : Fail with status(unsupported command)

andy31415, 2022-06-02:
@sowmyassp  what was the command executed? `mandatory attribute throws error` is somewhat vague. Could you also share the chip-tool commands used or at least the attribute names (at least one of them)

andy31415, 2022-06-02:
Test plans may change. It would be useful to see list of commands in the issue itself so that we can reference them in case test plans are updated.

andy31415, 2022-06-02:
https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/all-clusters-common/all-clusters-app.matter#L4246 seems to claim that actionList and endpointList are available as a callback:

```
server cluster BridgedActions {
    callback attribute actionList;
    callback attribute endpointList;
    callback attribute setupUrl;
    ram      attribute featureMap;
    callback attribute clusterRevision default = 1;
}
```

guessing callback probably not properly implemented or missing.

andy31415, 2022-06-02:
I also wonder if we should use all-clusters to test bridge as we have a dedicated bridge example.

sowmyassp, 2022-06-02:
@andy31415 I actually run this commands in all-clusters-app, once I will run in bridge app if I face same issue I will raise an issue then now I will close this ticket, thank you

closed 2022-06-02 by sowmyassp