project-chip/connectedhomeip #19068: [TC-DM-3.4] - API's to generate invalid conditions when sending UpdateNOC

sumaky, 2022-06-01:
Following are the requirements to validate TC-DM-3.4
API's to generate  NOC with different cases of 
- [ ] keyadded CAT tag
- [ ] changed nodeID
- [ ] changed subject public 
- [ ] NOC that is not in the TrustedRootCertificates
- [ ] NOC with invalid length

woody-apple, 2022-06-02:
Cert Blocker Review: @emargolis is working on this with @tcarmelveilleux.

franck-apple, 2022-06-21:
is there more work to be done here in addition to #19629, or can we close this issue?

sumaky, 2022-06-23:
@franck-apple API's for following are required
1.We could not see any IPKValue  int logs with trace enabled. 
AddNOC command requires IPKValue as its parameter.
2. If this issue is closed is readme file update as to how we can inject errors?
3.  from the help file we see the below commands we could not see IPK
no-error                         - No error to inject.
           cert-oversized                   - Certificate size will exceed it's muximum supported size, which is
                                              400 bytes for the CHIP TLV encoded cert and 600 bytes for DER encoded cert.
           cert-version                     - Certificate version will be set to v2 instead of required v3.
           serial-number-missing            - Certificate won't have required serialNumber field.
           sig-algo                         - Use ecdsa-with-SHA1 signature algorithm instead of required ecdsa-with-SHA256.
           issuer-missing                   - Certificate won't have required Issuer field.
           validity-not-before-missing      - Certificate won't have required validity not-before field.
           validity-not-after-missing       - Certificate won't have required validity not-after field.
           validity-wrong                   - Certificate will have validity not-before and not-after values switched,
                                              where not-before will have greater value than not-after.
           subject-missing                  - Certificate won't have required Subject field.
           subject-node-id-missing          - Subject won't have NodeId attribute.
           subject-node-id-invalid          - Subject will include invalid NodeId value.
           subject-node-id-twice            - Subject will include two NodeId attributes.
           subject-fabric-id-missing        - Subject won't have FabricId attribute.
           subject-fabric-id-invalid        - Subject will include invalid FabricId value.
           subject-fabric-id-twice          - Subject will include two FabricId attributes.
           subject-fabric-id-mismatch       - The FabricId in the subject won't match FabricId in the issuer field.
           subject-cat-invalid              - Subject will include invalid CASE Authenticated Tag (CAT)
 value.
           sig-curve                        - Use secp256k1 curve to generate certificate signature instead of
                                              required secp256r1 (aka prime256v1).
           publickey                        - Error will be injected in one of the bytes of the public key value.
                                              required secp256r1 (aka prime256v1).
           ext-basic-missing                - Certificate won't have required Basic Constraint extension.
           ext-basic-critical-missing       - Basic Constraint extension won't have critical field.
           ext-basic-critical-wrong         - Basic Constraint extension will be marked as non-critical.
           ext-basic-ca-missing             - Basic Constraint extension won't have cA field.
           ext-basic-ca-wrong               - Basic Constraint extension cA field will be set to TRUE for DAC
                                              and to FALSE for PAI and PAA.
           ext-basic-pathlen-presence-wrong - Basic Constraint extension will include pathLen field for NOC.
           ext-basic-pathlen0               - Basic Constraint extension pathLen field will be set to 0.
           ext-basic-pathlen1               - Basic Constraint extension pathLen field will be set to 1.
           ext-basic-pathlen2               - Basic Constraint extension pathLen field will be set to 2.
           ext-key-usage-missing            - Certificate won't have required Key Usage extension.
           ext-key-usage-critical-missing   - Key Usage extension won't have critical field.
           ext-key-usage-critical-wrong     - Key Usage extension will be marked as non-critical.
           ext-key-usage-dig-sig            - Key Usage extension digitalSignature flag won't be set for NOC
                                              and will be set for ICAC/RCAC.
           ext-key-usage-key-cert-sign      - Key Usage extension keyCertSign flag will be set for NOC
                                              and won't be set for ICAC/RCAC.
           ext-key-usage-crl-sign           - Key Usage extension cRLSign flag will be set for NOC
                                              and won't set for ICAC/RCAC.
           ext-akid-missing                 - Certificate won't have required Authority Key ID extension.
           ext-skid-missing                 - Certificate won't have required Subject Key ID extension.
           ext-extended-key-usage-missing   - Certificate won't have required Extended Key Usage extension.
           signature                        - Error will be injected in one of the bytes of the signature value.

cjandhyala, 2022-06-28:
Cert Blocker Review Update 06/28: @sumaky and team working Evgeni to generate invalid opcerts

emargolis, 2022-07-18:
There is no need to inject error into certificate when we need to generate NOC that doesn't chain up to the TrustedRootCertificates. In this case we just use different RootCertificate to sign ICAC.

Also, I don't think I understand the test cases listed in this ticket:
  - keyadded CAT tag
  - changed nodeID
  - changed subject public
These require clarifications.

tcarmelveilleux, 2022-07-27:
We need to have access to the root store of chip-tool for the chip-cert commands to properly generate these cert chains to be valid with UpdateNOC. I am working on Python method to do this test

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

cjandhyala, 2022-08-04:
Hi @tcarmelveilleux, Could you share any updates on the python script?

woody-apple, 2022-08-17:
Given this is now marked provisional in the linked test plan, moving out of 1.0

closed 2023-03-12 by stale[bot]