project-chip/connectedhomeip #19069: [pre-sve] TC-SU-2.7 not getting the right StateTransition event notification

Shadi-Silabs, 2022-06-01:
#### Problem
TC-SU-2.7: step#3 is not producing the right event notification. 

1. This has been tested on chip-tool, ota provider and end-device app all based on PRE-SVE week1 commit id but it failed.
2. The same has been tested on chip-tool, ota provider and end-device app all based on te9 commit id and it passed.
3. the same has been tested on a combination of chip-tool and ota are built based on te9 commit while end-device on PRE-SVE commit id, and it passed.
please check all the logs below
 

 #### Proposed Solution
it seems that something is wrong from the controller side. it is either from chip-tool or ot-provider.
[ota and chip on te9 while end-device on pre-sve.zip](https://github.com/project-chip/connectedhomeip/files/8818046/ota.and.chip.on.te9.while.end-device.on.pre-sve.zip)
[Pre-SVE logs.zip](https://github.com/project-chip/connectedhom
[te9.zip](https://github.com/project-chip/connectedhomeip/files/8818054/te9.zip)
eip/files/8818047/Pre-SVE.logs.zip)

ukatira, 2022-06-02:
Looks like timeout for chip-tool event subscribe is not working and session ends right away. Works for me when tried using interactive mode with these steps:

chip-tool interactive start
subscribe-event state-transition 5 10 0x1234567890 0
otasoftwareupdaterequestor announce-ota-provider 0x00000000ABCD 0 0 0 0x0000001234567890 0

cjandhyala, 2022-06-02:
@Shadi-Silabs Could you run this test in Interactive mode as Utasav was suggesting above.

Shadi-Silabs, 2022-06-02:
@cjandhyala @ukatira it worked with interactive mode. thank you

bzbarsky-apple, 2022-06-02:
subscribe commands are only supported in chip-tool in interactive mode...

closed 2022-06-02 by bzbarsky-apple