project-chip/connectedhomeip #19081: ESP32 platfomr GetClock_RealTime broken

msandstedt, 2022-06-02:
#### Problem

In init:

```
CHIP_ERROR InitClock_RealTime()
{
    Clock::Microseconds64 curTime =
        Clock::Microseconds64((static_cast<uint64_t>(CHIP_SYSTEM_CONFIG_VALID_REAL_TIME_THRESHOLD) * UINT64_C(1000000)));
    // Use CHIP_SYSTEM_CONFIG_VALID_REAL_TIME_THRESHOLD as the initial value of RealTime.
    // Then the RealTime obtained from GetClock_RealTime will be always valid.
    return System::SystemClock().SetClock_RealTime(curTime);
}
```

This is not the right thing to do.  This makes the platform believe and report that it does have wall clock time, even defeating its own rudimentary check in the getter:

```
CHIP_ERROR ClockImpl::GetClock_RealTime(Microseconds64 & aCurTime)
{
    struct timeval tv;
    if (gettimeofday(&tv, nullptr) != 0)
    {   
        return CHIP_ERROR_POSIX(errno);
    }
    if (tv.tv_sec < CHIP_SYSTEM_CONFIG_VALID_REAL_TIME_THRESHOLD)
    {   
        return CHIP_ERROR_REAL_TIME_NOT_SYNCED;
    }
    if (tv.tv_usec < 0)
    {   
        return CHIP_ERROR_REAL_TIME_NOT_SYNCED;
    }
    static_assert(CHIP_SYSTEM_CONFIG_VALID_REAL_TIME_THRESHOLD >= 0, "We might be letting through negative tv_sec values!");
    aCurTime = Microseconds64((static_cast<uint64_t>(tv.tv_sec) * UINT64_C(1000000)) + static_cast<uint64_t>(tv.tv_usec));
    return CHIP_NO_ERROR;
}
```

Because the clock is initialized to CHIP_SYSTEM_CONFIG_VALID_REAL_TIME_THRESHOLD (the chip epoch), the system will never discover that it does not have wall clock time and can never properly bubble up the CHIP_ERROR_REAL_TIME_NOT_SYNCED error.

 #### Proposed Solution

At a bare minimum, the platform would need to always return CHIP_ERROR_UNSUPPORTED_CHIP_FEATURE from its GetClock_RealTime methods to be spec compliant.

Better would be to set a flag that can capture whether the clock really has been set.

woody-apple, 2022-06-15:
Spec Issue Review: @msandstedt and @tcarmelveilleux. It sounds like there may be an SDK workaround here for 1.0. If so, is this still a critical issue for 1.0?

msandstedt, 2022-06-24:
@woody-apple ,

Yes, tthe platforms in the sdk that were returning obviously incorrect time have been patched to always return an error from their time-getter interfaces.  This may be acceptable for 1.0.

andy31415, 2022-07-12:
Removed spec for now (I assume this is not spec) and 1.0 as well based on latest comment from @msandstedt

closed 2023-01-20 by stale[bot]