project-chip/connectedhomeip #19091: TC-DM-3.1- updates to be made to the chip-cert tool to run test case

sumaky, 2022-06-02:
### The following are the requirements to complete the validation process for TC-DM-3.1 

1. CertificateChainResponse does not show the certificate data as shown below
CertificateChainResponse: {
[1651570197.289898][3689:3694] CHIP:TOO:     certificate: Elided value too large of size 491
[1651570197.289954][3689:3694] CHIP:TOO:    }
2. API to know the length of the certificate
3. CSRResponse Command fields has to be displayed in readable format for validation

- NOCSR Elements
- Attestation Signature

4 Generate AddTrustedRootCertificate Command( certificate data visible)
5.Generate Generate invalid root certificate- size exceeding 400 octstr
6. Update chip-cert tool to generate different mal-formatted root certs
7. Parse the TLV data (obtained from the console log) and print certificate (root, NoC, ICAC) for the tester to compare.

TC-DM-3.3
The reference all-cluster-app implementation has all these checks

bzbarsky-apple, 2022-06-02:
>  [1651570197.289898][3689:3694] CHIP:TOO: certificate: Elided value too large of size 491

You're using an old version of chip-tool, it looks like.  This was fixed in #18738 over a week ago.

> CSRResponse Command fields has to be displayed in readable format for validation

I believe https://github.com/project-chip/connectedhomeip/pull/18893 addresses this, but @krypton36 should know for sure.

> 4 Generate AddTrustedRootCertificate Command( certificate data visible)
>  5.Generate Generate invalid root certificate- size exceeding 400 octstr
> 6. Update chip-cert tool to generate different [...]

Again, @krypton36 should know the state of things here.

woody-apple, 2022-06-03:
Cert Blocker Review: Assigning to myself to break this up into tasks.

closed 2022-06-07 by woody-apple