project-chip/connectedhomeip #19095: TC-DA-1.3- updates to be made to the chip-cert tool to run test case

sumaky, 2022-06-02:
Need to add log statements in the reference app to validate the NONCE

woody-apple, 2022-06-03:
Cert Blocker Review: Assigning to @franck-apple to follow up here. This needs more information.

krypton36, 2022-06-06:
This is already logged in the app logs. See Examples Below:
Interactive Commands to :
`./out/host/chip-tool interactive start `
`pairing code-paseonly 1234 MT:-24J0YXE00KA0648G00`
`generalcommissioning arm-fail-safe  500 137438953472 1234 0`
`generalcommissioning set-regulatory-config 0 'us' 137438953472 1234 0`
`operationalcredentials certificate-chain-request 1 1234 0`

## Cmd + Accessory Log:
```
operationalcredentials attestation-request hex:955D0A79044D301B28D015D3007D852DDC8748D57DA8682A6BD8276478A9B60F 1234 0
[1654541323.897627][18211:18211] CHIP:DMG: 			CommandData = 
[1654541323.897708][18211:18211] CHIP:DMG: 			{
[1654541323.897785][18211:18211] CHIP:DMG: 				0x0 = [
[1654541323.897892][18211:18211] CHIP:DMG: 					0x95, 0x5d, 0xa, 0x79, 0x4, 0x4d, 0x30, 0x1b, 0x28, 0xd0, 0x15, 0xd3, 0x0, 0x7d, 0x85, 0x2d, 0xdc, 0x87, 0x48, 0xd5, 0x7d, 0xa8, 0x68, 0x2a, 0x6b, 0xd8, 0x27, 0x64, 0x78, 0xa9, 0xb6, 0xf, 
[1654541323.897981][18211:18211] CHIP:DMG: 			]
```

## Cmd + Accessory Log:
```
operationalcredentials csrrequest hex:2F5E32456CEE7C66E676871D27A5CBD5FBEEB178E620ECBFBCB579109980E97D 1234 0
[1654541330.621702][18211:18211] CHIP:DMG: 			CommandData = 
[1654541330.621783][18211:18211] CHIP:DMG: 			{
[1654541330.621860][18211:18211] CHIP:DMG: 				0x0 = [
[1654541330.621968][18211:18211] CHIP:DMG: 					0x2f, 0x5e, 0x32, 0x45, 0x6c, 0xee, 0x7c, 0x66, 0xe6, 0x76, 0x87, 0x1d, 0x27, 0xa5, 0xcb, 0xd5, 0xfb, 0xee, 0xb1, 0x78, 0xe6, 0x20, 0xec, 0xbf, 0xbc, 0xb5, 0x79, 0x10, 0x99, 0x80, 0xe9, 0x7d, 
[1654541330.622055][18211:18211] CHIP:DMG: 			]
```

And with [PR](https://github.com/project-chip/connectedhomeip/pull/18893) you will get the following log:
```
Attestation Nonce (32) = 955D0A79044D301B28D015D3007D852DDC8748D57DA8682A6BD8276478A9B60F
```

krypton36, 2022-06-13:
@sumaky  The Attestation None and the CSR Nonce can be obtained with the following command:
`echo hex:$(hexdump -vn32 -e'4/4 "%08X" ' /dev/urandom)`

This can be done twice for each command.

sumaky, 2022-06-13:
> I would assume fixed, however this bug may need to be open to make sure test plans mention what portion of the log to use to validate the nonce (the `Attestation Nonce` line)

@andy31415 Test plan a in a way generic. I am not sure if we could have specifics of logs in test plans. We will for sure update the Verification Sheet with the commands used.

sumaky, 2022-06-14:
We have verified this issue. We are able to see the nonce. Thanks.

closed 2022-06-14 by sumaky