project-chip/connectedhomeip #19098: TC-DA-1.4-updates to be made to the chip-cert tool to run test case

sumaky, 2022-06-02:
### **Step1 :Generated the invalid vectors** 
examples

- Invalid NOCSR Information 
- CSR is not of the type OCTET string 
- CSRNonce is OCTET string of length > 32 
- NOCSRElement size is greater than RESP_MAX 
- AttestationSignature is not an octet string
### **Step2:**
Inject the invalid vectors into the device and test for the error conditions.

woody-apple, 2022-06-03:
Cert Blocker Review: Marking cert blocker, assigning to @krypton36

cjandhyala, 2022-06-03:
@krypton36 is going to give instruction to @sumaky on how to run these tests using all-cluster-app. @sumaky to verify and get back

krypton36, 2022-06-06:
## Validate Cert Decleration:
### Build Cert Tool
1. `gn gen --check out/host`

2. `ninja -v -C out/host chip-cert `
### Generate Cert Declaration
1. ```./out/host/chip-cert gen-cd -C ./credentials/development/paa-root-certs/Chip-Test-PAA-FFF1-Cert.pem -K credentials/development/attestation/Chip-Development-PAA-Key.pem -O Dec_Message.txt -f 1 -V FFF1 -p 8000 -p 8001 -p 8002 -p 8003 -p 8004 -p 8005 -p 8006 -p 8007 -p 8008 -p 8009 -p 800A -p 800B -p 800C -p 800D -p 800E -p 800F -p 8010 -p 8011 -p 8012 -p 8013 -p 8014 -p 8015 -p 8016 -p 8017 -p 8018 -p 8019 -p 801A -p 801B -p 801C -p 801D -p 801E -p 801F -p 8020 -p 8021 -p 8022 -p 8023 -p 8024 -p 8025 -p 8026 -p 8027 -p 8028 -p 8029 -p 802A -p 802B -p 802C -p 802D -p 802E -p 802F -p 8030 -p 8031 -p 8032 -p 8033 -p 8034 -p 8035 -p 8036 -p 8037 -p 8038 -p 8039 -p 803A -p 803B -p 803C -p 803D -p 803E -p 803F -p 8040 -p 8041 -p 8042 -p 8043 -p 8044 -p 8045 -p 8046 -p 8047 -p 8048 -p 8049 -p 804A -p 804B -p 804C -p 804D -p 804E -p 804F -p 8050 -p 8051 -p 8052 -p 8053 -p 8054 -p 8055 -p 8056 -p 8057 -p 8058 -p 8059 -p 805A -p 805B -p 805C -p 805D -p 805E -p 805F -p 8060 -p 8061 -p 8062 -p 8063 -d 0016 -c "ZIG20142ZB330003-24" -l 0 -i 0 -n 2694 -t 0 ```
2. `xxd -p -c 1024 Dec_Message.txt `

### Use CD in chip-all-clusters-app
1. Copy Result from Step 2 of [Gen CD](#generate-cert-declaration) and place it in [dac_provider_example.txt](https://github.com/project-chip/connectedhomeip/files/8847006/dac_provider_example.txt) JSON file.
2. Transfer the updated JSON file to Raspi That will be running chip-all-clusters-app
3. ./out/host/chip-all-clusters-app --discriminator 3840 --dac_provider ~/dac_example.json 
4. In Step 1 of [Gen CD](#generate-cert-declaration) the values can be updated as needed for the test steps. See `--help` of the `chip-cert` tool for error conditions. 
### DAC Scenarios
1. See this [README](https://github.com/project-chip/connectedhomeip/tree/master/src/tools/chip-cert#chip-certificate-tool) for DAC generation and see `--help` of the` chip-cert `tool for error scenarios.
2. Use `xxd -p -c 1024` on the DER generated file of the cert to get the cert payload.
3. Private key can be obtain by doing `xxd -p -c 1024` on the DER file for private key.

franck-apple, 2022-06-22:
Per conversation with @sumaky , closing this issue. Will re-open if necessary.

closed 2022-06-22 by franck-apple