project-chip/connectedhomeip #19108: [pre-sve] TC-ULABEL-2.2 unable to write label to label-list attribute

doublemis1, 2022-06-02:
#### Problem
SHA: 425f1fb916ca154deeb75cd82cd5196778aba152
platform: nrf
example: all cluster app
Failure in response from write label-list. 
./chip-tool userlabel write label-list '[{"label":"roomName", "value":"master bedroom 1"}, {"label":"orientation", "value":"east"}, {"label":"floor", "value":"2"}, {"label":"roomType", "value":"bedroom"}]' 1 0

```
[1654178684.766214][99957:99962] CHIP:DMG: 		AttributeStatusIB =
[1654178684.766216][99957:99962] CHIP:DMG: 		{
[1654178684.766218][99957:99962] CHIP:DMG: 			AttributePathIB =
[1654178684.766221][99957:99962] CHIP:DMG: 			{
[1654178684.766223][99957:99962] CHIP:DMG: 				Endpoint = 0x0,
[1654178684.766226][99957:99962] CHIP:DMG: 				Cluster = 0x41,
[1654178684.766228][99957:99962] CHIP:DMG: 				Attribute = 0x0000_0000,
[1654178684.766231][99957:99962] CHIP:DMG: 				ListIndex = Null,
[1654178684.766233][99957:99962] CHIP:DMG: 			}
[1654178684.766236][99957:99962] CHIP:DMG: 				
[1654178684.766238][99957:99962] CHIP:DMG: 			StatusIB =
[1654178684.766240][99957:99962] CHIP:DMG: 			{
[1654178684.766243][99957:99962] CHIP:DMG: 				status = 0x01 (FAILURE),
[1654178684.766245][99957:99962] CHIP:DMG: 			},
[1654178684.766248][99957:99962] CHIP:DMG: 			
[1654178684.766250][99957:99962] CHIP:DMG: 		},
```

[chip_tool.log](https://github.com/project-chip/connectedhomeip/files/8824630/chip_tool.log)

woody-apple, 2022-06-07:
Cert Blocker Review: Removing cert blocker, given the Core SDK supports this, but does need resolution in the specific platform.

Damian-Nordic, 2022-06-09:
Looks like Andy has already add initialization of DeviceInfoProvider for all nRF Connect apps: https://github.com/project-chip/connectedhomeip/pull/18847.