project-chip/connectedhomeip #19113: Git issue cyw30739_sdk

dtodor, 2022-06-02:
It seems that the `cyw30739_sdk` references unavailable/inaccessible commits:

```
fatal: remote error: upload-pack: not our ref ab7601a74bdffc7374353c1e19b22388bb3dcedd
Fetched in submodule path 'third_party/cyw30739_sdk/repos/30739A0', but it did not contain ab7601a74bdffc7374353c1e19b22388bb3dcedd. Direct fetching of that commit failed.
```

dtodor, 2022-06-02:
OK, had to manually change the remote URLs to:

```
https://github.com/Infineon/OT-Matter-30739A0.git
https://github.com/Infineon/OT-Matter-TARGET_CYW930739M2EVB-01.git
```

vringar, 2022-06-03:
The change should have happened with https://github.com/project-chip/connectedhomeip/pull/18733  but somehow didn't

XuXiangJun, 2022-06-06:
I have the same problem, is there any temporary solution?

vringar, 2022-06-06:
You need to change the `.gitmodules` file.
I have the fix local rn.
I'll create a PR in a bit

vringar, 2022-06-06:
https://github.com/project-chip/connectedhomeip/pull/19240

vringar, 2022-06-06:
Apparently the two repo links have been fixed, as I don't see them in the diff. However the matter authors have also used tags in the `branches` field, which didn't work for me, so I changes those to proper branches instead.

XuXiangJun, 2022-06-15:
> You don't have to change the remote URLs or `.gitmodules` manually. Please use `git submodule sync` and `git submodule update` to update submodules.

It works. Thank you very much.
`git submodule sync --recursive`
`git submodule update --init --recursive`

closed 2022-12-20 by stale[bot]