project-chip/connectedhomeip #19116: lighting-app has occupancy sensing server cluster

fessehaeve, 2022-06-02:
#### Problem

According to spec OnOffLight device type has occupancy sensing client cluster as optional. Having the server cluster here seems wrong.

 #### Proposed Solution
Replace occupancy sensing server with client.

bzbarsky-apple, 2022-06-03:
If this is done, Binding also needs to be added on the endpoint, since it will then have a client cluster.

jvmahon, 2024-01-07:
It seems odd to have an optional Occupancy Sensing cluster for On/Off Light type 0x0100 and Dimmable Light 0x0101. Shouldn't Occupancy Sensing just be done as a separate endpoint?  Also seems odd to include it in the Pump type as a Optional cluster rather than as a separate endpoint.

Note that it also seems inconsistent (why is it on those two light types but not Color Temperature Light 0x010C and Extended Color Light 0x010D).  Seems that implementing as a separate endpoint when needed would be more consistent.