project-chip/connectedhomeip #19132: Some commands can take longer in sync execution time than the MRP timeout

bzbarsky-apple, 2022-06-02:
#### Problem
Some commands take longer than the MRP timeout to execute.  Their execution then blocks standalone acks from being sent, and the sender keeps resending.

We have run into this with AddNOC/UpdateNOC (due to the crypto bits they have to do).

I believe @bluebin14 has also run into it for some commands that touched storage when storage was being (due to a bug) very slow.  But in general, we don't have a good idea of how long sync storage might take.

 #### Proposed Solution
For commands that involve known slow crypto, we should probably just add manual `FlushAcks()` calls.

For others, not clear.

bzbarsky-apple, 2022-06-02:
Also, the newly added IM timeout of 2 seconds + MRP bits is probably too low for these too... @mrjerryjohns @erjiaqing

erjiaqing, 2022-06-02:
> Also, the newly added IM timeout of 2 seconds + MRP bits is probably too low for these too... @mrjerryjohns @erjiaqing 

The caller can still set the timeout for each command they send.

The 2s timeout should cover 99% cases. Since only cryptographic bits and some async commands are using longer time.

bluebin14, 2022-06-03:
Although the storage fragmentation bug has been fixed, back then only an MRP of 5s+2s on both client and server could make it reliably past commissioning (changing it on one side had no effect). Also, RemoveFabric was just as slow as AddNOC/UpdateNOC because it has to do a lot of cleanup.

bzbarsky-apple, 2022-07-02:
This is going to somewhat change what's happening on the wire (not at the protocol level, but in terms of actual messages sent), when fixed...

woody-apple, 2022-07-05:
SDK Review: Given this will make the test harness more resilient to a variety of DUTs, we believe this is appropriate for the SVE branch.

commit c3cd004e (2022-07-08): Fix UpdateNOC session invalidation and opcreds timing (#20461)

* Fix UpdateNOC session invalidation and opcreds timing

- UpdateNOC did not clear session of previous fabric like spec intended
  (#20379)
- All OpCreds cluster slow commands did not try to early-ack to avoid
  MRP timeouts (#19132)

Fixes #20379
Fixes #19132

This PR:

- Fixes UpdateNOC expiring all sessions for the updated node
- Adds `FlushAcksRightAwayOnSlowCommand` to CommandHandler to flush acks
  on slow commands
- Adds Python tests for UpdateNOC behavior of session expiring
- Adds `ExpireSessions` to Python for testing

T [... cut]
PR #20512 (2022-07-09): Fix UpdateNOC session invalidation and opcreds timing
#### Problem
- UpdateNOC did not clear session of previous fabric like spec intended
  (#20379)
- All OpCreds cluster slow commands did not try to early-ack to avoid
  MRP timeouts (#19132)

Fixes #20379
Fixes #19132

#### Change overview
- Fixes UpdateNOC expiring all sessions for the updated node
- Adds `FlushAcksRightAwayOnSlowCommand` to CommandHandler to flush acks
  on slow commands
- Adds Python tests for UpdateNOC behavior of session expiring
- Adds `ExpireSessions` to Python for testing

#### Testing
- Unit tests all pass
- Cert te [... cut]
commit a6470a08 (2022-07-09): Fix UpdateNOC session invalidation and opcreds timing (#20461) (#20512)

* Fix UpdateNOC session invalidation and opcreds timing

- UpdateNOC did not clear session of previous fabric like spec intended
  (#20379)
- All OpCreds cluster slow commands did not try to early-ack to avoid
  MRP timeouts (#19132)

Fixes #20379
Fixes #19132

This PR:

- Fixes UpdateNOC expiring all sessions for the updated node
- Adds `FlushAcksRightAwayOnSlowCommand` to CommandHandler to flush acks
  on slow commands
- Adds Python tests for UpdateNOC behavior of session expiring
- Adds `ExpireSessions` to Python for t [... cut]
closed 2022-07-08 by tcarmelveilleux