project-chip/connectedhomeip #19135: Pairing of M5 board via CHIPTool cli throws Timeout error

phonnakasturi-apple, 2022-06-02:
With SHA 6c4b54ec544cf748180c1e32a41f6e4b1b34b7fe

M5 board cannot be paired via chiptool cli.

**Steps to reproduce the issue;**

Note: Ensure that M5 board is connected and flashed

1. cd ~/connectedhomeip
2. git reset --hard 6c4b54ec544cf748180c1e32a41f6e4b1b34b7fe
3. source scripts/bootstrap.sh
4. source scripts/activate.sh
5. gn gen out/debug --args='chip_mdns="platform" chip_crypto="mbedtls" chip_inet_config_enable_ipv4=false'
6. ninja -C out/debug
7. sudo ./out/debug/chip-tool pairing ble-wifi 4 "<wifi name>" <password> 20202021 3840 --paa-trust-store-path ./credentials/development/paa-root-certs

**Error:**

[1654206140364] [89910:2429897] CHIP: [EM] OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:265: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1654206140421] [89910:2429898] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:205648405 on exchange 43505i
[1654206140421] [89910:2429898] CHIP: [EM] Found matching exchange: 43505i, Delegate: 0x7fcd34004250
[1654206140421] [89910:2429898] CHIP: [DMG] ICR moving to [ResponseRe]
[1654206140421] [89910:2429898] CHIP: [DMG] InvokeResponseMessage =
[1654206140421] [89910:2429898] CHIP: [DMG] {
[1654206140421] [89910:2429898] CHIP: [DMG] 	suppressResponse = false, 
[1654206140421] [89910:2429898] CHIP: [DMG] 	InvokeResponseIBs =
[1654206140421] [89910:2429898] CHIP: [DMG] 	[
[1654206140421] [89910:2429898] CHIP: [DMG] 		InvokeResponseIB =
[1654206140421] [89910:2429898] CHIP: [DMG] 		{
[1654206140421] [89910:2429898] CHIP: [DMG] 			CommandDataIB =
[1654206140421] [89910:2429898] CHIP: [DMG] 			{
[1654206140421] [89910:2429898] CHIP: [DMG] 				CommandPathIB =
[1654206140421] [89910:2429898] CHIP: [DMG] 				{
[1654206140421] [89910:2429898] CHIP: [DMG] 					EndpointId = 0x0,
[1654206140421] [89910:2429898] CHIP: [DMG] 					ClusterId = 0x30,
[1654206140421] [89910:2429898] CHIP: [DMG] 					CommandId = 0x1,
[1654206140421] [89910:2429898] CHIP: [DMG] 				},
[1654206140421] [89910:2429898] CHIP: [DMG] 				
[1654206140421] [89910:2429898] CHIP: [DMG] 				CommandFields = 
[1654206140421] [89910:2429898] CHIP: [DMG] 				{
[1654206140421] [89910:2429898] CHIP: [DMG] 					0x0 = 0, 
[1654206140421] [89910:2429898] CHIP: [DMG] 					0x1 = "", 
[1654206140421] [89910:2429898] CHIP: [DMG] 				},
[1654206140421] [89910:2429898] CHIP: [DMG] 			},
[1654206140421] [89910:2429898] CHIP: [DMG] 			
[1654206140421] [89910:2429898] CHIP: [DMG] 		},
[1654206140421] [89910:2429898] CHIP: [DMG] 		
[1654206140421] [89910:2429898] CHIP: [DMG] 	],
[1654206140421] [89910:2429898] CHIP: [DMG] 	
[1654206140421] [89910:2429898] CHIP: [DMG] 	InteractionModelRevision = 1
[1654206140421] [89910:2429898] CHIP: [DMG] },
[1654206140422] [89910:2429898] CHIP: [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0001
[1654206140422] [89910:2429898] CHIP: [CTL] Failsafe disarmed
[1654206140422] [89910:2429898] CHIP: [CTL] Successfully finished commissioning step 'Cleanup'
[1654206140422] [89910:2429898] CHIP: [TOO] Device commissioning Failure: ../../src/app/CommandSender.cpp:211: CHIP Error 0x00000032: Timeout
[1654206140422] [89910:2429898] CHIP: [DIS] Closing all BLE connections
[1654206140422] [89910:2429898] CHIP: [IN] Clearing BLE pending packets.
[1654206140422] [89910:2429898] CHIP: [BLE] Auto-closing end point's BLE connection.
[1654206140422] [89910:2429898] CHIP: [DMG] ICR moving to [AwaitingDe]
[1654206140422] [89910:2429893] CHIP: [CTL] Shutting down the commissioner
[1654206140422] [89910:2429893] CHIP: [CTL] Shutting down the controller
[1654206140422] [89910:2429893] CHIP: [CTL] Shutting down the commissioner
[1654206140423] [89910:2429893] CHIP: [CTL] Shutting down the controller
[1654206140423] [89910:2429893] CHIP: [IN] Expiring all connections for fabric 1!!
[1654206140423] [89910:2429893] CHIP: [CTL] Shutting down the commissioner
[1654206140423] [89910:2429893] CHIP: [CTL] Shutting down the controller
[1654206140423] [89910:2429893] CHIP: [IN] Expiring all connections for fabric 2!!
[1654206140423] [89910:2429893] CHIP: [CTL] Shutting down the commissioner
[1654206140423] [89910:2429893] CHIP: [CTL] Shutting down the controller
[1654206140423] [89910:2429893] CHIP: [IN] Expiring all connections for fabric 3!!
[1654206140423] [89910:2429893] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1654206140423] [89910:2429893] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654206140423] [89910:2429893] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654206140423] [89910:2429893] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654206140423] [89910:2429893] CHIP: [DMG] IM WH moving to [Uninitialized]
[1654206140423] [89910:2429893] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1654206140423] [89910:2429893] CHIP: [BLE] CancelConnection
[1654206140423] [89910:2429893] CHIP: [DL] Inet Layer shutdown
[1654206140423] [89910:2429893] CHIP: [DL] BLE shutdown
[1654206140423] [89910:2429893] CHIP: [DL] System Layer shutdown
[1654206140423] [89910:2429893] CHIP: [IN] SecureSession Released 0x7fcd32482a80 Type:1 LSID:10578
[1654206140423] [89910:2429893] CHIP: [TOO] Run command failure: ../../src/app/CommandSender.cpp:211: CHIP Error 0x00000032: Timeout

**Attaching the complete log file;**

[M5-cli-error-log.txt](https://github.com/project-chip/connectedhomeip/files/8828094/M5-cli-error-log.txt)

bzbarsky-apple, 2022-06-02:
Here is the relevant part of the log:
```
[1654206138037] [89910:2429898] CHIP: [CTL] Performing next commissioning step 'SendNOC'
...
[1654206138037] [89910:2429898] CHIP: [CTL] Sent operational certificate to the device
...
[1654206140040] [89910:2429898] CHIP: [DMG] Time out! failed to receive invoke command response from Exchange: 43504i
[1654206140040] [89910:2429898] CHIP: [CTL] Device failed to receive the operational certificate Response: ../../src/app/CommandSender.cpp:211: CHIP Error 0x00000032: Timeout
```

Note that the AddNOC command timed out in 2 seconds, because that's the timeout that #18744 set.  There's no way an m5stack can do an AddNOC in 2 seconds.  I just tried commissioning an m5stack on the changeset right before that PR merged and here's what a successful log looks like:

```
[1654210729427] [72482:3413866] CHIP: [CTL] Sent operational certificate to the device
...
[1654210732363] [72482:3413865] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:92935151 on exchange 20798i
```
which is 2936 ms to get the response.

That said, why did we time out in 2s?  I would have expected the exchange timeout to be set to "2s plus however long it takes to do all our MRP resends", not a flat 2s.

@mrjerryjohns @erjiaqing why is the code from #18744 not at least accounting for MRP bits here and doing a hard 2 second cutoff?

bzbarsky-apple, 2022-06-02:
Ah, this is BLE, so no MRP involved.  So that's why the timeout is a flat 2s.

bzbarsky-apple, 2022-06-03:
It looks like for TCP we assume 30s for packet retransmits and whatnot, while for BLE we assume 0.

bzbarsky-apple, 2022-06-03:
Fix coming up.

woody-apple, 2022-06-03:
Cert Blocker Review: Marking cert blocker.

closed 2022-06-03 by woody-apple