project-chip/connectedhomeip #19139: [TC-IDM-3.1]Chip-tool support needed to send a WriteRequest message from TH to DUT to write any attribute on two clusters.

Rajashreekalmane, 2022-06-03:
#### Problem
As per Test-Plan, test-step 15 says "DUT sends a WriteRequestMessage to the DUT to both the clusters with the appropriate dataversions(received in the previous step) to modify the value of an attribute ",
Chip-tool support needed to send a  WriteRequestMessage  from DUT  to TH to read any attribute on two clusters.

**PFA log below:-**
[image]

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#ref_WriteRequestAction

bzbarsky-apple, 2022-06-03:
Sounds like this would need multi-path write support in chip-tool...

woody-apple, 2022-06-03:
Cert Blocker Review: Marking cert blocker, assigning to @vivien-apple.

vivien-apple, 2022-06-09:
Do we actually supports multiple writes in a single invoke in the SDK ? I thought it was not a v1.0 thing.
@mrjerryjohns @woody-apple ?

bzbarsky-apple, 2022-06-09:
> Do we actually supports multiple writes in a single invoke in the SDK ?

We support multiple writes in a single write request message, yes.  We do not support multiple _commands_ in a single _invoke_.

vivien-apple, 2022-06-14:
> > Do we actually supports multiple writes in a single invoke in the SDK ?
> 
> We support multiple writes in a single write request message, yes. We do not support multiple _commands_ in a single _invoke_.

I was wrong as you said. Supports is in #19139

PR #19566 (2022-06-15): [chip-tool] Add multiple writes in a single transaction support
#### Problem

`chip-tool` does not supports multiple writes in a single interaction.

#### Change overview
 * Add supports to multiple writes in a single interaction.
    * `./out/debug/standalone/chip-tool onoff write on-time 13 0x12344321 1,2,3 `
    * `./out/debug/standalone/chip-tool onoff write-by-id 0x4001, 0x4001 '19',17' 0x12344321 1`
    * `./out/debug/standalone/chip-tool any write-by-id 0x6,0x7 0x4001,0x0 '5','false' 0x12344321 1,1`

fixes #19139
closed 2022-06-15 by andy31415