project-chip/connectedhomeip #19140: Lighting-app example missing Binding Cluster

pankore, 2022-06-03:
#### Problem
According to spec, Binding Cluster is part of the Base Cluster Requirements and is mandatory for all device types.
Lighting-app example is missing this Binding Cluster

 #### Proposed Solution
Add binding cluster to lighting-app

bzbarsky-apple, 2022-06-03:
> According to spec, Binding Cluster is part of the Base Cluster Requirements and is mandatory for all device types.

Where do you see that @pankore ?

"1.1.7. Base Cluster Requirements for Matter" in the device library says it's required for all device types _that include a client cluster_ and are simple device types.  Lighting app implements the "Dimmable Light" device type, for which the only client cluster is an optional "Occupancy Sensing" client.  And lighting app does not implement that client....  

That said, if https://github.com/project-chip/connectedhomeip/issues/19116 is fixed, then it would indeed need Binding.

pankore, 2022-06-03:
Noted, closing this issue.

closed 2022-06-03 by pankore