project-chip/connectedhomeip #19146: [pre-sve] [TC-CC-1.1] FeatureMap AttributeList mismatch

jimlyall-q, 2022-06-03:
#### Problem
The feature map is 31 which indicates full features
The attribute list only has 26 entries
 * expected behavior is steps 3 and 4 being consistent with instructions
 * See logs for steps to reproduce
 * tc-cc-1.1
 * x64 CTRL + qpg6105
 * screenshots, images, logs, etc. [TC_CC_1_1.1654192596.log](https://github.com/project-chip/connectedhomeip/files/8831347/TC_CC_1_1.1654192596.log)

 >

 #### Proposed Solution
None

jimlyall-q, 2022-06-03:
Causes a lot of tests to be failing e.g. TC-CC-2-1 feature map has XY attribute doesn't

bzbarsky-apple, 2022-06-03:
@jmartinez-silabs can you take a look, please?

jmartinez-silabs, 2022-06-03:
I am not sure to understand the issue with the attribute list or your mention of

> TC-CC-2-1 feature map has XY attribute doesn't

How many attributes is the test expecting?

Here is the attribute list. Note I added the hex value next to them
```
[1654273525.827957][2563:2569] CHIP:TOO:   AttributeList: 26 entries
[1654273525.827988][2563:2569] CHIP:TOO:     [1]: 0  (0x00)
[1654273525.828013][2563:2569] CHIP:TOO:     [2]: 1  (0x01)
[1654273525.828038][2563:2569] CHIP:TOO:     [3]: 2 (0x02)
[1654273525.828062][2563:2569] CHIP:TOO:     [4]: 3 (0x03)
[1654273525.828086][2563:2569] CHIP:TOO:     [5]: 4 (0x04)
[1654273525.828109][2563:2569] CHIP:TOO:     [6]: 7 (0x07)
[1654273525.828133][2563:2569] CHIP:TOO:     [7]: 8 (0x08)
[1654273525.828157][2563:2569] CHIP:TOO:     [8]: 15 (0x0F)
[1654273525.828181][2563:2569] CHIP:TOO:     [9]: 16 (0x10)
[1654273525.828205][2563:2569] CHIP:TOO:     [10]: 16384 (0x4000)
[1654273525.828229][2563:2569] CHIP:TOO:     [11]: 16385 (0x4001)
[1654273525.828253][2563:2569] CHIP:TOO:     [12]: 16386 (0x4002)
[1654273525.828277][2563:2569] CHIP:TOO:     [13]: 16387 (0x4003)
[1654273525.828301][2563:2569] CHIP:TOO:     [14]: 16388 (0x4004)
[1654273525.828325][2563:2569] CHIP:TOO:     [15]: 16389 (0x4005)
[1654273525.828349][2563:2569] CHIP:TOO:     [16]: 16390 (0x4006)
[1654273525.828372][2563:2569] CHIP:TOO:     [17]: 16394  (0x400a)
[1654273525.828396][2563:2569] CHIP:TOO:     [18]: 16395 (0x400b)
[1654273525.828420][2563:2569] CHIP:TOO:     [19]: 16396 (0x400c)
[1654273525.828444][2563:2569] CHIP:TOO:     [20]: 16397 (0x400d)
[1654273525.828468][2563:2569] CHIP:TOO:     [21]: 16400 (0x4010)
[1654273525.828491][2563:2569] CHIP:TOO:     [22]: 65528 (0xFFF8)
[1654273525.828515][2563:2569] CHIP:TOO:     [23]: 65529 (0xFFF9)
[1654273525.828650][2563:2569] CHIP:TOO:     [24]: 65531 (0xFFFB)
[1654273525.828696][2563:2569] CHIP:TOO:     [25]: 65532 (0xFFFC)
[1654273525.828735][2563:2569] CHIP:TOO:     [26]: 65533 (0xFFFD)
```
Which one is missing?

https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/colorcontrol.adoc#test-procedure
Step 4

https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/ColorControl.adoc#17-color-information-attribute-set

All mandatory or featured need attributes and commands are there and expected by the test plan. Also, XY is supported by the light app. What test fails and what is it expecting?

cjandhyala, 2022-06-03:
@jimlyall-q , What is the app you are using to test this, on raspi all-cluster-app we get 56 entries. I believe this needs to be looked from the app that is run on your platform.

franck-apple, 2022-06-07:
Marking it `cert blocker`, as it is already tagged `cert blocker request` and it is a pre-SVE issue.

jimlyall-q, 2022-06-08:
We're running the lighting-app as the all-clusters-app is too large to run on our platform.
Just highlighting the fact that the instructions should be more generic or that there should be options to test against?

yinyihu-silabs, 2022-06-08:
@jimlyall-q The attribute list you're getting is correct. those are all the mandatory attributes.
the ones you see in the validation steps with 57 attribute entries, many of them are optional attributes.
for example: 2, 5, 6 are optional
and the ones below are also all optional attributes which does not depends on feature-map but rather number-of-primaries or other factors.

> [1649661013.012109][7948:7953] CHIP:TOO:     [12]: 17
[1649661013.012134][7948:7953] CHIP:TOO:     [13]: 18
[1649661013.012168][7948:7953] CHIP:TOO:     [14]: 19
[1649661013.012194][7948:7953] CHIP:TOO:     [15]: 21
[1649661013.012228][7948:7953] CHIP:TOO:     [16]: 22
[1649661013.012253][7948:7953] CHIP:TOO:     [17]: 23
[1649661013.012286][7948:7953] CHIP:TOO:     [18]: 25
[1649661013.012311][7948:7953] CHIP:TOO:     [19]: 26
[1649661013.012346][7948:7953] CHIP:TOO:     [20]: 27
[1649661013.012371][7948:7953] CHIP:TOO:     [21]: 32
[1649661013.012406][7948:7953] CHIP:TOO:     [22]: 33
[1649661013.012430][7948:7953] CHIP:TOO:     [23]: 34
[1649661013.012465][7948:7953] CHIP:TOO:     [24]: 36
[1649661013.012490][7948:7953] CHIP:TOO:     [25]: 37
[1649661013.012523][7948:7953] CHIP:TOO:     [26]: 38
[1649661013.012548][7948:7953] CHIP:TOO:     [27]: 40
[1649661013.012573][7948:7953] CHIP:TOO:     [28]: 41
[1649661013.012606][7948:7953] CHIP:TOO:     [29]: 42
[1649661013.012631][7948:7953] CHIP:TOO:     [30]: 48
[1649661013.012655][7948:7953] CHIP:TOO:     [31]: 49
[1649661013.012679][7948:7953] CHIP:TOO:     [32]: 50
[1649661013.012703][7948:7953] CHIP:TOO:     [33]: 51
[1649661013.012726][7948:7953] CHIP:TOO:     [34]: 52
[1649661013.012750][7948:7953] CHIP:TOO:     [35]: 54
[1649661013.012774][7948:7953] CHIP:TOO:     [36]: 55
[1649661013.012798][7948:7953] CHIP:TOO:     [37]: 56
[1649661013.012821][7948:7953] CHIP:TOO:     [38]: 58
[1649661013.012845][7948:7953] CHIP:TOO:     [39]: 59
[1649661013.012869][7948:7953] CHIP:TOO:     [40]: 60

jimlyall-q, 2022-06-14:
Is it possible to have a result that can be tested against for the light/lock app?

closed 2022-06-14 by jimlyall-q