project-chip/connectedhomeip #19149: [pre-sve] [TC-CC-3.1] move-to-hue doesn't move

jimlyall-q, 2022-06-03:
#### Problem
TC-CC-3.1 Steps 1 thru 2b

 * expected behavior - after issuing command and waiting at least TransitionTime seconds value changes from 0 to 1
 * actual behavior - no change
 * steps to reproduce - TC-CC-3.1 Steps 1 thru 2b
 * system configuration - RPi +qpg6105
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
Could probably use

jimlyall-q, 2022-06-03:
[TC_CC_3_2.1654238195.log](https://github.com/project-chip/connectedhomeip/files/8831847/TC_CC_3_2.1654238195.log)
Could be the same here with move-hue

jimlyall-q, 2022-06-03:
The values aren't moving in the updated instructions
https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit#gid=531994392

For TC-CC-5.1
(step 3b)1649665872.712086 - (step 2)1649665828.577879 ~= 44s which is 10% of transitionTime would expect a change in value

bzbarsky-apple, 2022-06-03:
@jimlyall-q Is the device in the "on" state or the "off" state when these steps are being executed?  The second test setup step in the test plan says to make sure it's in "on" but those test setup steps are not in the spreadsheet you linked.  @cjandhyala 

The "log files" linked above seem to be HTML, not log files.

closed 2022-12-11 by stale[bot]