project-chip/connectedhomeip #19150: [TC-MC-1.3] AttributeList and AcceptedCommandList value mismatch error

manjunath-grl, 2022-06-03:
Summary:
As per Testplan [TC-MC-1.3] and spec
1. "AttributeList" expected value: [0,1,65528, 65529, 65531, 65532, 65533] but got value: [0,65528, 65529, 65531, 65532, 65533] 
2. "AcceptedCommandList" expected value: [0, 1, 2] but got value: [0]
when executed 
Testplan screenshot:
[image]
[image]
Spec screenshot:
[image]
[image]
Execution logs:
[image]
[MC1_3_ACL.txt](https://github.com/project-chip/connectedhomeip/files/8831855/MC1_3_ACL.txt)
[image]
[MC1_3_AL.txt](https://github.com/project-chip/connectedhomeip/files/8831859/MC1_3_AL.txt)

bzbarsky-apple, 2022-06-03:
> AttributeList" expected value: [0,1,65528, 65529, 65531, 65532, 65533]

The attribute with id 1 is optional.  Why is the test plan requiring it to be present?  @manjunath-grl @cjandhyala

woody-apple, 2022-06-03:
Cert Blocker Review: Assigning to @franck-apple to follow up to see if we can update the test plans here.

franck-apple, 2022-06-03:
after discussion with @cjandhyala , assigning to @manjunath-grl for more information.

manjunath-grl, 2022-06-08:
@bzbarsky-apple @franck-apple 1. AttributeList : we checked with QA team it is  passing for them we used same app and endpoint but we are observing missing Optional attribute, we are waiting on logs to investigate further 
2. AcceptedCommandList : commands are mandatory we got [0] but the expected value is [0,1,2] as per spec
Spec screenshot:
[image]

manjunath-grl, 2022-06-13:
@bzbarsky-apple @franck-apple tried in TOT its passing
1. SHA used: c1e1d018820c0fdd51f124f607b9207f31fae2ff
2. Client app: Tv-app
3. Server app: Chip-tool
[TC_MC_1_3.zip](https://github.com/project-chip/connectedhomeip/files/8890129/TC_MC_1_3.zip)

closed 2022-06-13 by manjunath-grl

bzbarsky-apple, 2022-06-13:
> Client app: Tv-app
> Server app: Chip-tool

I assume those are backwards...