project-chip/connectedhomeip #19151: Audit examples/darwin-framework-tool/templates/tests/partials/checks/maybeCheckExpectedConstraints.zapt

vivien-apple, 2022-06-03:
Should this have a check for optionality and being missing if optional?  I think it should...

As in, we should not have "is actual value nil" checks scattered about with specific constraints.  For example, for an optional thing a minLength constraint should pass if not present, but the code below will treat the actual length as 0, I think...

I guess this just got moved; can you please file a followup to fix this up?

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/19042#discussion_r887000551_

vivien-apple, 2022-06-03:
I have also noticed something that looks broken for string handling when I moved the code at examples/darwin-framework-tool/templates/tests/partials/checks/maybeCheckExpectedConstraints.zapt 

It is unclear to me what `asTypedLiteral` is here for strings.

bzbarsky-apple, 2022-06-10:
asTypedLiteral is broken for strings, but it's broken in the same way for chip-tool, fwiw...  `notValue` constraints for strings only work if the `notValue` points to a variable.

closed 2022-06-13 by bzbarsky-apple