project-chip/connectedhomeip #19152: [TC-MC-1.6] AttributeList and AcceptedCommandList value mismatch error

manjunath-grl, 2022-06-03:
Summary:
As per testplan [TC-MC-1.6] 
"AttributeList" expected value: [0,1,2,65528, 65529, 65531, 65532, 65533] but got value: [0,65528, 65529, 65531, 65532, 65533]
"AcceptedCommandList" expected value: [0, 2, 3] but got value: [0,2]
when executed
Testplan screenshot:
[image]
[image]
Spec screenshot:
[image]
[image]
Execution logs:
[image]
[MC1_6_ACL.txt](https://github.com/project-chip/connectedhomeip/files/8831947/MC1_6_ACL.txt)
[image]
[MC1_6_AL.txt](https://github.com/project-chip/connectedhomeip/files/8831949/MC1_6_AL.txt)

chrisdecenzo, 2022-06-03:
hi @manjunath-grl  What version of the example/tv-app were you using? Eg. was this latest TOT or TE9 or something else? When I run with TOT, I see all the attributes for these two clusters so trying to figure out what might be different in your environment

manjunath-grl, 2022-06-06:
@chrisdecenzo i tested it in latest TOT against Tv-app in Linux system

chrisdecenzo, 2022-06-07:
hi @manjunath-grl I see the top line in the txt files is:

~/CHIP/APR13/connectedhomeip

Can you please double check that you have mainline synced to TOT because these missing commands/attributes were fixed in a PR from May timeframe

chrisdecenzo, 2022-06-07:
Specifically, this is fixed by https://github.com/project-chip/connectedhomeip/pull/18762

closed 2022-06-07 by chrisdecenzo