project-chip/connectedhomeip #19154: [TC-MC-1.7] AttributeList value mismatch error

manjunath-grl, 2022-06-03:
Summary:
As per Testplan [TC-MC-1.7] and spec "AttributeList" expected value: [0,1,2,3,4,5,665528, 65529, 65531, 65532, 65533] but got value: [0,1,2,4,5,6,65528, 65529, 65531, 65532, 65533] value 3 (SampledPosition attribute) missing.

Testplan screenshot:
[image]
Spec screenshot:
[image]
Execution logs:
[image]
[MC1_7_AL.txt](https://github.com/project-chip/connectedhomeip/files/8832009/MC1_7_AL.txt)

andy31415, 2022-06-03:
@manjunath-grl - what is the environment being tested? what app and what platform? https://github.com/project-chip/connectedhomeip/blob/master/docs/BUG_REPORT.md#environment-for-reproduction

chrisdecenzo, 2022-06-03:
hi @manjunath-grl  What version of the example/tv-app were you using? Eg. was this latest TOT or TE9 or something else? When I run with TOT, I see all the attributes for these two clusters so trying to figure out what might be different in your environment

manjunath-grl, 2022-06-06:
@chrisdecenzo i tested it in latest TOT against Tv-app in Linux system

chrisdecenzo, 2022-06-07:
Specifically, this is fixed by https://github.com/project-chip/connectedhomeip/pull/18762

closed 2022-06-07 by chrisdecenzo