project-chip/connectedhomeip #19159: [TC-OO-2.3] showing value mismatch error for some of the steps

manjunath-grl, 2022-06-03:
Summary:
As per testplan [TC-OO-2.3] attribute OnOff,OffWaitTime and OnTime fails for some steps showing value mismatch error.
1.Step 11b:  OffWaitTime expected value 300 but got 299
2.Step 18b: 1.OffWaitTime expected value 300 but got 0
                   2.OnOff expected value 1 but got 0
                   3.OnTime expected value 300 but got 0
3.Step 19c: 1.OffWaitTime expected value 300 but got 299
                    2.OnTime expected value 200 but got 201
4.Step 20b: OffWaitTime expected value 300 but got 299
5.Step 21b: OffWaitTime expected value 200 but got 300
Testplan screenshot:
[image]
[image]
[image]
[image]
[image]
Spec screenshot:
[image]
Execution logs:
[image]
[11b_OffWaitTime.txt](https://github.com/project-chip/connectedhomeip/files/8832252/11b_OffWaitTime.txt)
[image]
[18b_OffWaitTime.txt](https://github.com/project-chip/connectedhomeip/files/8832254/18b_OffWaitTime.txt)
[18b_OnOff.txt](https://github.com/project-chip/connectedhomeip/files/8832257/18b_OnOff.txt)
[18b_OnTime.txt](https://github.com/project-chip/connectedhomeip/files/8832259/18b_OnTime.txt)
[19c_OnTime.txt](https://github.com/project-chip/connectedhomeip/files/8832261/19c_OnTime.txt)
[20b_OffWaitTime.txt](https://github.com/project-chip/connectedhomeip/files/8832263/20b_OffWaitTime.txt)
[21b_OffWaitTime.txt](https://github.com/project-chip/connectedhomeip/files/8832264/21b_OffWaitTime.txt)

bzbarsky-apple, 2022-06-03:
>  1.Step 11b: OffWaitTime expected value 300 but got 299

@manjunath-grl The time starts off as 300, but time has a tendency to pass, so by the time it's read it may be less than 300.  Note that the expectation in the test plan explicitly says "approximately equal".

For step 18, @jmartinez-silabs could you take a look please?

jmartinez-silabs, 2022-06-03:
I closed the same issue yesterday
https://github.com/project-chip/connectedhomeip/issues/18943

I will copy my comment and my result here. But this is strictly not respecting the instructions or timings

> Hi all, I went over the spec, test-plan and ran the Test case TC-OO-2.3. The implementation and instructions are correct. This is not an issue nor a cert blocker.
> 
> I have attached the results of the test I ran here
> [TC-OO-2-3.txt](https://github.com/project-chip/connectedhomeip/files/8827984/TC-OO-2-3.txt)
> 
> You have to be careful with the command and timing throughout the steps. Some command turns off the light after the 

> given time/or prevent the light from being turned back on from the previously given offWaitTime and then some commands use the parameter AcceptOnlyWhenOn set to 1 which mean the command will be ignored if the OnOff state is off.
> 
> For example, if you wait too long between Steps 9b and 10a, the Ontime from step 9 will expire, the light will be off and the command from 10a will be successful but ignored. 10b and 10c will then fail. If that is the case you need to turn the light back on before doing step 10a.

bzbarsky-apple, 2022-06-03:
@manjunath-grl @cjandhyala Can we get whatever is going on here with the test script that causes it to be confused fixed, please?

woody-apple, 2022-06-03:
Cert Blocker Review: Marking cert blocker, assigning to @franck-apple to follow up about updating the test plan/yaml.

franck-apple, 2022-06-04:
Per discussion with @cjandhyala , assigning to @manjunath-grl to see how QA is passing this test

manjunath-grl, 2022-06-08:
@RajashreeKalmane can you please send me the manual run logs of this test.

somu1710, 2022-06-14:
@manjunath-grl 

There is no issue with test plan or test verification steps.

Please do check the expected outcome carefully and proceed - https://docs.google.com/spreadsheets/d/1h4d4J90KAxFqUL_cXZHyM3pHjHCEXzFITkQhN2KMqDY/edit?pli=1#gid=1886169525&range=B102:G104

I am attaching the logs : [TC-OO-2.3.txt](https://github.com/project-chip/connectedhomeip/files/8899863/TC-OO-2.3.txt)

Please verify and close the issue.

manjunath-grl, 2022-08-10:
Fixed in [PR](https://github.com/CHIP-Specifications/chip-test-plans/issues/2024) closing the ticket

closed 2022-08-10 by manjunath-grl