project-chip/connectedhomeip #19163: [pre-sve] TC-WNCV-2.4 - wrong type attribute based on FeatureMap value

doublemis1, 2022-06-03:
#### Problem
all cluster app
nrf52840
revision: https://github.com/project-chip/connectedhomeip/commit/425f1fb916ca154deeb75cd82cd5196778aba152

1 |   | WNCV.S.A0000(Type) | TH reads Type attribute from DUT | Value has to be between a valid range* of [min=0, max=9] and If (LF & !TL) value must be between [min=0, max=5] or equals to 9, and If (!LF & TL) value must be only equal to 7, and If (LF & TL) value is either 6 or 8
-- | -- | -- | -- | --

The Type value should be set to 6 or 8 while the device supports PA and LA.
Output from DUT:
```
[1654263036.637398][91610:91615] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0102 Attribute 0x0000_0000 DataVersion: 3973067745
[1654263036.637432][91610:91615] CHIP:TOO:   Type: 0
```
Device featureMap
```
[1654263285.591000][92229:92234] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0102 Attribute 0x0000_FFFC DataVersion: 3973067745
[1654263285.591052][92229:92234] CHIP:TOO:   FeatureMap: 23
```
10111 -> support PA_TL, PA_LF, LF, TL

bzbarsky-apple, 2022-06-03:
@jmeg-sfy can you take a look, please?

jmeg-sfy, 2022-06-03:
> @jmeg-sfy can you take a look, please?
@bzbarsky-apple ok thanks

woody-apple, 2022-06-03:
Cert Blocker Review: Marking cert blocker, based on reference platforms.

LuDuda, 2022-06-08:
Sure will handle this, thanks.

PR #19311 (2022-06-09): [window-app] Set valid Type for Window Covering cluster
#### Problem
If Feature Map of Window Covering cluster has both LF and TL bits set, Type should be either 6 or 8. 

#### Change overview
Initialize the attribute to 8 in window-app and all-clusters-app so that they can pass certification tests.
Fixes #19163

#### Testing
Tested manually using nRF Connect all-clusters-app
closed 2022-06-09 by Damian-Nordic